cyanfish/naps2 #36: 7.0b1 Scanning now opens extra windows, can hang

jcurl, 2023-01-02:
**Describe the bug**
This appears to be a regression over v6.1.2. An extra window appears when using my TWAIN scanner (that did not appear before). It is visible with a short title bar only. Closing this window closes the scanner TWAIN interface and NAPS is no longer responsive. Must go to the task manager to close.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Scan -> Choose Scanner.
[image]
2. The scanner now appears on the bottom right. Next to it is an empty window
[image]
3. Closing that window and NAPS now hangs

The profile window for the scanner is:
[image]

[image]

**Expected behavior**
Same behaviour as in NAPS 6.1.2

**Desktop (please complete the following information):**
 - Windows 10 19043.

**Additional context**
The same occurs for my OKI scanner also. It might be related to recent changes in TWAIN?

cyanfish, 2023-01-03:
I'll try and fix the extra window later. But in 7.0b2 I put a change that might help with the hang - please try and let me know if it changes anything.

jcurl, 2023-01-04:
Still problematic in NAPS 7.0b2.

closed 2023-01-08 by cyanfish

cyanfish, 2023-01-08:
Thanks, I think this should be fixed in [7.0b3](https://www.naps2.com/download). Please let me know if it works for you.

commit a726c205 (2023-01-08): Improve twain window management

Fixes #36