cyanfish/naps2 #37: Importing recently saved document fails

jcurl, 2023-01-02:
**Describe the bug**
Scanned in a document using NAPS2 7.01b. Saved it as a PDF-1a. Importing the document new results in an exception that it could not be imported.

**To Reproduce**
Steps to reproduce the behavior:
1. Scan in a document, save it.
2. Click on Import, choose the PDF that was recently saved.
3. See error
```
System.IO.IOException: Error reading pdf file '\...\file.pdf'.
   at NAPS2.ImportExport.Pdf.PdfImporter.LoadDocument(String filePath, ImportParams importParams)
   at NAPS2.ImportExport.Pdf.PdfImporter.<>c__DisplayClass7_0.<Import>b__0(Action`1 produceImage)
   at NAPS2.Util.AsyncProducers.<>c__DisplayClass0_0`1.<RunProducer>b__0(Action`1 produce)
   at NAPS2.Util.AsyncProducers.<>c__DisplayClass1_0`1.<<RunProducer>b__0>d.MoveNext()
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at NAPS2.Util.AsyncSink`1.<AsAsyncEnumerable>d__3.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at System.Threading.Tasks.Sources.ManualResetValueTaskSourceCore`1.GetResult(Int16 token)
   at NAPS2.ImportExport.ImportOperation.<>c__DisplayClass2_0.<<Start>b__0>d.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at NAPS2.ImportExport.ImportOperation.<>c__DisplayClass2_0.<<Start>b__0>d.MoveNext()
```

I don't believe the document is in use by any other application (for this, it should have returned AccessDenied).

**Expected behavior**
Either a clear error description on what is wrong on how it can be resolved, or that it works.

**Screenshots**
None applicable.

**Desktop (please complete the following information):**
 - Windows 10.19043 64-bit

**Additional context**
None.

cyanfish, 2023-01-02:
I can reproduce this when there are unicode characters in the path. I'll include a fix in the next beta.

closed 2023-01-03 by cyanfish

jcurl, 2023-01-03:
Thank you. I removed the name for privacy reasons. It does have an umlaut `ä` on a UNC path under Windows.

jcurl, 2023-01-04:
Problem still occurs. I'll create a new ticket.

commit 8fddf04c (2023-01-03): Fix importing PDFs with unicode paths

Fixes #37