cyanfish/naps2 #38: Regression, drag and drop of PDF no longer works

jcurl, 2023-01-02:
**Describe the bug**
In NAPS 6.1.2 it was possible to drag and drop a PDF onto the main window, and it would be imported. With NAPS 7.0b1 it is no longer possible.

**To Reproduce**
Steps to reproduce the behavior:
1. Find a PDF document
2. Drag and drop it onto the Window from NAPS, it is empty (just started, no existing document).
3. The icon is marked that drag and drop is not possible, and releasing is no action.

**Expected behavior**
Drag and drop of the PDF should import the PDF.

**Screenshots**
No screenshot possible while drag and drop in progress.

**Desktop (please complete the following information):**
 - Windows 10.19043 64-bit

**Additional context**
None.

bebobebo, 2023-01-03:
Can't drag and drop image files e.g. PNG to import either

closed 2023-01-03 by cyanfish

cyanfish, 2023-01-03:
Thanks, this should be fixed in [beta 2](https://www.naps2.com/download).

jcurl, 2023-01-04:
Drag and drop now works, but not if there is a NON-ASCII char. e.g. if the file is called `foobär.pdf` it raises an exception. See https://github.com/cyanfish/naps2/issues/49

commit b075977c (2023-01-03): WinForms: Fix drag/drop

Fixes #38