cyanfish/naps2 #39: Feature Request: Touch Up Tool

jcurl, 2023-01-02:
**Is your feature request related to a problem? Please describe.**
When often scanning in documents, there are artifacts, such as black lines, or dots, that I'd like to remove. Having a square select tool that I can just press delete, which makes the text white would be helpful in cleaning up the document before saving. This would be helpful to remove page edges, punch holes and other artifacts for archiving.

**Describe the solution you'd like**
When one enters the individual PDF pages, where deskewing, cropping, etc. can occur, my suggestion would be to have a select tool that can be used to select regions on the scanned pdf. When selecting, pressing DEL would erase the text to be white.

**Describe alternatives you've considered**
The only alternative I see is to export them all as images, use editors to do the "minor" changes to the document and then reimport. This workflow is about 10x more work than if it were integrated into the program.

**Additional context**
None

christianclavet, 2023-04-28:
This is one thing I would like to add. A new tool to "mask" image. The first iteration would use a rectangle fill (Using almost the same code as the crop tool) with the color sampling icon to get the color of the page to mask the artifact. Then later would like to add a small brush to do some small touch-up masking. Could add a gradient border using alpha so it blend well with the page.

I'm not there yet, and need to adapt my ideas from my current fork (from 6.1.2) to this new source to propose Pull Requests...

cyanfish, 2025-06-08:
In [NAPS2 8.2.0](https://www.naps2.com/download) you can now select "Edit with" under the image menu to do whatever editing you want in Paint or your preferred image editor.

closed 2025-06-08 by cyanfish