cyanfish/naps2 #41: 7.0b1 Scanning and then saving first page incorrect size

jcurl, 2023-01-02:
**Describe the bug**
When saving a document, the first page is not the correct size, but seems much larger.

I scan using a scanner that has automatic feed, and it is told to scan at a standard size of A4. This always worked in NAPS 6.1.2. In the PDF view, it appears to be good. After saving (I have OCR for German enabled, and PDF/A-1b is set), and then opening in Adobe Acrobat DC reader, the first page is larger than it should be.

**To Reproduce**
Steps to reproduce the behavior:
1. Scan in a document with multiple pages, with the resolution set by the scanning software (Canon DR-C240) to A4.
2. Observe all pages, perhaps deskew them. The document has already had automatic OCR.
3. Save the document as a PDF
   * I retested without PDF/A and the problem exists there too.
4. Open the document using PDF reader like Adobe Acrobat DC. The first page is the wrong size
   * interestingly, when clicking on the page blank area where it normally selects the entire page in Adobe Reader, it choose the A4 size subset, not the complete page.
   * The subsequent pages appear to be the correct size.

**Expected behavior**
The pages saved in the PDF document should be the same sizes as it was scanned. 

**Screenshots**
Here's a screenshot of the most recent document (redacted). You'll see the top half is all white with the full width (larger than A4), where the second page (reflow in Adobe enabled) is the correct size.

[image]

**Desktop (please complete the following information):**
 - Windows 10 64-bit 19043

**Additional context**
Add any other context about the problem here.

cyanfish, 2023-01-04:
Do you have the same problem if you save to a JPG image?

Can you reproduce with a non-sensitive PDF that you can upload here?

cyanfish, 2023-01-04:
And one more question - if you reorder the pages before saving, does the problem stick with the page as you move it, or is it always the first page in the PDF?

jcurl, 2023-01-04:
I think this is a bug / regression in Adobe PDF Reader. I noticed yesterday that other documents from my Bank are now distorted.

What is interesting still, is whatever the bug in Adobe, NAPS 6.1.2 doesn't tickle it, but 7.0b1 did.

Here's an A4 color document. Both pages should be the same, but Adobe Reader seems to have problems. I checked using Foxit Reader and it works OK. The NAPS shows the correct size.

This is what I see with Adobe:
[image]
Please note the right part of the top page is cropped white in this case.

This is what Foxit shows
[image]

Here's the scanned in PDF.
[TestPage.pdf](https://github.com/cyanfish/naps2/files/10343697/TestPage.pdf)

Because I've come to the conclusion this is an Adobe Bug (old documents that worked seem to have problems now), I'm closing it.

closed 2023-01-04 by jcurl