cyanfish/naps2 #42: 6.1.2 Can't download/install OCR languages

jcurl, 2023-01-02:
**Describe the bug**
After reinstalling 6.1.2 I want to add German to my OCR languages. I select it, it downloads three files, but after downloading it indicates an error.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to OCR
2. Choose German
3. Click OK and follow the prompts.

**Expected behavior**
Downloading should work.

**Screenshots**
None

**Desktop (please complete the following information):**
 - Windows 10 x64 19043

**Additional context**
It doesn't do this on all computers. I noticed an "errorlog.txt". Here's the output.

```text
2023-01-02 13:00:17.9832 Error preparing downloaded file System.ArgumentException: 1 is not a supported code page.
Parameter name: codepage
   at System.Text.Encoding.GetEncoding(Int32 codepage)
   at ICSharpCode.SharpZipLib.Zip.ZipFile.ReadEntries()
   at ICSharpCode.SharpZipLib.Zip.ZipFile..ctor(String name)
   at NAPS2.Dependencies.DownloadFormat.ZipDownloadFormat.Extract(String zipFilePath, String outDir)
   at NAPS2.Dependencies.DownloadFormat.ZipDownloadFormat.Prepare(String tempFilePath)
   at NAPS2.WinForms.FDownloadProgress.StartNextDownload()
2023-01-02 13:00:20.6056 Error preparing downloaded file System.ArgumentException: 1 is not a supported code page.
Parameter name: codepage
   at System.Text.Encoding.GetEncoding(Int32 codepage)
   at ICSharpCode.SharpZipLib.Zip.ZipFile.ReadEntries()
   at ICSharpCode.SharpZipLib.Zip.ZipFile..ctor(String name)
   at NAPS2.Dependencies.DownloadFormat.ZipDownloadFormat.Extract(String zipFilePath, String outDir)
   at NAPS2.Dependencies.DownloadFormat.ZipDownloadFormat.Prepare(String tempFilePath)
   at NAPS2.WinForms.FDownloadProgress.StartNextDownload()
2023-01-02 13:00:24.5930 Error preparing downloaded file System.ArgumentException: 1 is not a supported code page.
Parameter name: codepage
   at System.Text.Encoding.GetEncoding(Int32 codepage)
   at ICSharpCode.SharpZipLib.Zip.ZipFile.ReadEntries()
   at ICSharpCode.SharpZipLib.Zip.ZipFile..ctor(String name)
   at NAPS2.Dependencies.DownloadFormat.ZipDownloadFormat.Extract(String zipFilePath, String outDir)
   at NAPS2.Dependencies.DownloadFormat.ZipDownloadFormat.Prepare(String tempFilePath)
   at NAPS2.WinForms.FDownloadProgress.StartNextDownload()
2023-01-02 13:15:56.0298 Error preparing downloaded file System.ArgumentException: 1 is not a supported code page.
Parameter name: codepage
   at System.Text.Encoding.GetEncoding(Int32 codepage)
   at ICSharpCode.SharpZipLib.Zip.ZipFile.ReadEntries()
   at ICSharpCode.SharpZipLib.Zip.ZipFile..ctor(String name)
   at NAPS2.Dependencies.DownloadFormat.ZipDownloadFormat.Extract(String zipFilePath, String outDir)
   at NAPS2.Dependencies.DownloadFormat.ZipDownloadFormat.Prepare(String tempFilePath)
   at NAPS2.WinForms.FDownloadProgress.StartNextDownload()
2023-01-02 13:15:58.8717 Error preparing downloaded file System.ArgumentException: 1 is not a supported code page.
Parameter name: codepage
   at System.Text.Encoding.GetEncoding(Int32 codepage)
   at ICSharpCode.SharpZipLib.Zip.ZipFile.ReadEntries()
   at ICSharpCode.SharpZipLib.Zip.ZipFile..ctor(String name)
   at NAPS2.Dependencies.DownloadFormat.ZipDownloadFormat.Extract(String zipFilePath, String outDir)
   at NAPS2.Dependencies.DownloadFormat.ZipDownloadFormat.Prepare(String tempFilePath)
   at NAPS2.WinForms.FDownloadProgress.StartNextDownload()
```

cyanfish, 2023-01-02:
This is a [known issue](https://sourceforge.net/p/naps2/tickets/941/) in 6.1.2, I believe it's related to using certain non-English languages for Windows.

But you shouldn't have to re-download after upgrading to 7.x and downgrading back to 6.x, that's an oversight on my part that I'll fix in the next beta.

jcurl, 2023-01-04:
Confirmed this is fixed in 7.0b2. Closing.

closed 2023-01-04 by jcurl