cyanfish/naps2 #43: Cannot create file - error when saving new PDF

bebobebo, 2023-01-03:
Windows 11 
Uninstalled naps2 6.x before installing this beta 7.0.1.0
Made no adjustments to out of box settings

1. imported 4 PNG files (each ~50kb, created by Paint.NET)
2. Clicked Save PDF, typed in file name, 

Error: An error occurred when trying to save the file:

System.AggregateException: One or more errors occurred. ---> System.IndexOutOfRangeException: Index was outside the bounds of the array.
   at PdfSharpCore.Pdf.Advanced.PdfImage.ReadTrueColorMemoryBitmap(Int32 components, Int32 bits, Boolean hasAlpha)
   at PdfSharpCore.Pdf.Advanced.PdfImage..ctor(PdfDocument document, XImage image)
   at PdfSharpCore.Pdf.Advanced.PdfImageTable.GetImage(XImage image)
   at PdfSharpCore.Pdf.PdfPage.GetImageName(XImage image)
   at PdfSharpCore.Drawing.Pdf.XGraphicsPdfRenderer.Realize(XImage image)
   at PdfSharpCore.Drawing.Pdf.XGraphicsPdfRenderer.DrawImage(XImage image, Double x, Double y, Double width, Double height)
   at PdfSharpCore.Drawing.XGraphics.DrawImage(XImage image, Double x, Double y, Double width, Double height)
   at NAPS2.ImportExport.Pdf.PdfExporter.DrawImageOnPage(PdfPage page, IMemoryImage image, ImageExportFormat exportFormat, PdfCompat compat)
   at NAPS2.ImportExport.Pdf.PdfExporter.WriteToPdfSharpStep(PageExportState state)
   at System.Threading.Tasks.Dataflow.TransformBlock`2.ProcessMessage(Func`2 transform, KeyValuePair`2 messageWithId)
   at System.Threading.Tasks.Dataflow.TransformBlock`2.<>c__DisplayClass9_0.<.ctor>b__3(KeyValuePair`2 messageWithId)
   at System.Threading.Tasks.Dataflow.Internal.TargetCore`1.ProcessMessagesLoopCore()
   --- End of inner exception stack trace ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at NAPS2.Threading.Pipeline.PipelineStep`1.<Run>d__10.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at NAPS2.ImportExport.Pdf.PdfExporter.<>c__DisplayClass2_0.<<Export>b__0>d.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at NAPS2.ImportExport.Pdf.PdfExporter.<Export>d__2.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at NAPS2.ImportExport.Pdf.SavePdfOperation.<>c__DisplayClass8_0.<<Start>b__1>d.MoveNext()
---> (Inner Exception #0) System.IndexOutOfRangeException: Index was outside the bounds of the array.
   at PdfSharpCore.Pdf.Advanced.PdfImage.ReadTrueColorMemoryBitmap(Int32 components, Int32 bits, Boolean hasAlpha)
   at PdfSharpCore.Pdf.Advanced.PdfImage..ctor(PdfDocument document, XImage image)
   at PdfSharpCore.Pdf.Advanced.PdfImageTable.GetImage(XImage image)
   at PdfSharpCore.Pdf.PdfPage.GetImageName(XImage image)
   at PdfSharpCore.Drawing.Pdf.XGraphicsPdfRenderer.Realize(XImage image)
   at PdfSharpCore.Drawing.Pdf.XGraphicsPdfRenderer.DrawImage(XImage image, Double x, Double y, Double width, Double height)
   at PdfSharpCore.Drawing.XGraphics.DrawImage(XImage image, Double x, Double y, Double width, Double height)
   at NAPS2.ImportExport.Pdf.PdfExporter.DrawImageOnPage(PdfPage page, IMemoryImage image, ImageExportFormat exportFormat, PdfCompat compat)
   at NAPS2.ImportExport.Pdf.PdfExporter.WriteToPdfSharpStep(PageExportState state)
   at System.Threading.Tasks.Dataflow.TransformBlock`2.ProcessMessage(Func`2 transform, KeyValuePair`2 messageWithId)
   at System.Threading.Tasks.Dataflow.TransformBlock`2.<>c__DisplayClass9_0.<.ctor>b__3(KeyValuePair`2 messageWithId)
   at System.Threading.Tasks.Dataflow.Internal.TargetCore`1.ProcessMessagesLoopCore()<---

closed 2023-01-03 by cyanfish

cyanfish, 2023-01-03:
Thanks, this should be fixed in [beta 2](https://www.naps2.com/download).

bebobebo, 2023-01-03:
Confirmed fix.

commit f00579ca (2023-01-03): Fix PNG PDF export for widths that are not word-aligned

Fixes #43