cyanfish/naps2 #44: PDF compression

regs01, 2023-01-03:
**Is your feature request related to a problem? Please describe.**
Files could be pretty large sometimes.

**Describe the solution you'd like**
Ghostscript offers some compression levels. Would be good to include them into PDF settings.

**Describe alternatives you've considered**

**Additional context**

AltoRetrato, 2023-02-03:
I also miss (a lot!) options to set the resolution, quality and type of image compression when creating PDF files. For now, I end up editing the PDF generated by NAPS2 on Xara, so I can set the bitmap compression (Native, None, JPEG, Flate), custom DPI and JPEG quality.

Even though I could at first set the resolution before scanning, this means sometimes having to scan the same document twice! I might want to batch scan several pages / images and save some as a JPEG with a large resolution, and after that save all the scanned images as a PDF - sometimes with large resolution and quality (when I'm archiving photos, research/work results, etc.), other times as a low res./low quality quick reference (invoices, receipts, etc.).

bovergaauw, 2024-08-20:
I agree, being able to set PDF quality parameters would be great. I currently circumvent the lack of such settings by saving as TIFF and converting to PDF, but for some family members that's a little too complicated...

AltoRetrato, 2024-08-20:
I don't use NAPS2 anymore (mostly due to this issue), so please excuse me if I'm stating the obvious or talking about something that is already implemented.

Here are some references for the developers (or anyone that wants to patch the software):

- [Optimizing PDFs with Ghostscript](https://artifex.com/blog/optimizing-pdfs-with-ghostscript) (blog post by Artifex, makers of Ghostscript)
- [Ghostscript documentation online - Distiller parameters](https://ghostscript.readthedocs.io/en/gs10.03.1/VectorDevices.html#distiller-parameters) - I believe it would be simple to at least allow the user to select one of the distiller parameters: screen, ebook, printer, prepress. This is usually available in many other programs using Ghostscript, and each of them provides a different level of quality vs compression.
- Optionally, users could also provide custom values for relevant parameters, such as *DownsampleType, *ImageResolution, ColorConversionStrategy, DownScaleFactor, QFactor, VSamples and HSamples. This would allow advanced control over some of Ghostscript's quality settings.
- Allowing users to include custom PostScript code (as using the `-c` command-line parameter with Ghostscript) could give users the most freedom and versatility to define parameters for Ghostscript, even though it might be too advanced for most users...

bovergaauw, 2024-08-20:
For as far as I'm concerned, just an option to select the PDF quality as it is done in IrfanView would be just fine.

KevinADiehl, 2026-09-01:
Hi @cyanfish , 

I wanted to bump this because the value of this feature becomes even greater when looking at how people use NAPS2 today.

NAPS2 is no longer just a tool for initiating fresh scans—it has evolved into a fantastic, lightweight utility for editing and managing existing PDFs via the import function. We use it to reorder pages, rotate layouts, and merge documents together. 

Adding a compression or quality adjustment slider during the Save phase would perfectly fill a missing slot in NAPS2's excellent utility toolbox. Right now, if we import a massive, uncompressed PDF to tweak a few pages, we have no way to optimize its file size upon export without relying on external third-party software or Ghostscript. 

This single addition would act as a universal catch-all: it would optimize freshly scanned batches and compressed/uncompressed imported PDFs right at the final step. It perfectly fills a missing slot in NAPS2's excellent utility toolbox, eliminating the need to rely on external third-party software just to downsize a file for an email.

Thanks for developing and maintaining such a great piece of software!

cyanfish, 2026-10-12:
In [8.5.0](https://www.naps2.com/download) there are various improvements to PDF sizes by default. Less OCR overhead, JBIG2 compression, more efficient JPEG compression. There's also a "Compress" function under the Image menu that lets you change image quality after scanning/importing.

closed 2026-10-12 by cyanfish

regs01, 2026-10-12:
Can't it be Ghostscript type compression? So to scan in full quality and then save in selected quality, as it could be saved in multiple formats, like all pages to pdf and selected pages into quality image.