cyanfish/naps2 #46: Faulting application name: NAPS2.exe, version: 6.1.2.25834 on Windows 11

PeteSral, 2023-01-03:
**Describe the bug**
Crashing after startup.  Attempted to reinstall, did not fix the issue.

**To Reproduce**
Steps to reproduce the behavior:
1. Starting the application on Windows 11
2. App disappears and closes a few seconds.

**Desktop (please complete the following information):**
 - OS: Windows 11 Pro
 - Version Edition: 	22H2 OS build	22621.963 with Experience Pack 1000.22638.1000.0
 - Display Settings: 1920x1200, scale 100%

**Additional context**
Event viewer errors
**(1 of 2)**
Faulting application name: NAPS2.exe, version: 6.1.2.25834, time stamp: 0x5d3c9614
Faulting module name: KERNELBASE.dll, version: 10.0.22621.963, time stamp: 0x766ca8ae
Exception code: 0xe0434352
Fault offset: 0x00000000000906bc
Faulting process id: 0x0x6014
Faulting application start time: 0x0x1D91F9016164637
Faulting application path: C:\Program Files (x86)\NAPS2\NAPS2.exe
Faulting module path: C:\WINDOWS\System32\KERNELBASE.dll
Report Id: 6934c597-d8c0-40ac-91f8-0a49c5f1cb35
Faulting package full name: 
Faulting package-relative application ID: 

**(2 of 2)**
Application: NAPS2.exe
Framework Version: v4.0.30319
Description: The process was terminated due to an unhandled exception.
Exception Info: System.ServiceModel.EndpointNotFoundException

Server stack trace: 
   at System.ServiceModel.Channels.ConnectionUpgradeHelper.DecodeFramingFault(ClientFramingDecoder decoder, IConnection connection, Uri via, String contentType, TimeoutHelper& timeoutHelper)
   at System.ServiceModel.Channels.ClientFramingDuplexSessionChannel.SendPreamble(IConnection connection, ArraySegment`1 preamble, TimeoutHelper& timeoutHelper)
   at System.ServiceModel.Channels.ClientFramingDuplexSessionChannel.DuplexConnectionPoolHelper.AcceptPooledConnection(IConnection connection, TimeoutHelper& timeoutHelper)
   at System.ServiceModel.Channels.ConnectionPoolHelper.EstablishConnection(TimeSpan timeout)
   at System.ServiceModel.Channels.ClientFramingDuplexSessionChannel.OnOpen(TimeSpan timeout)
   at System.ServiceModel.Channels.CommunicationObject.Open(TimeSpan timeout)
   at System.ServiceModel.Channels.ServiceChannel.OnOpen(TimeSpan timeout)
   at System.ServiceModel.Channels.CommunicationObject.Open(TimeSpan timeout)
   at System.ServiceModel.Channels.ServiceChannel.CallOpenOnce.System.ServiceModel.Channels.ServiceChannel.ICallOnce.Call(ServiceChannel channel, TimeSpan timeout)
   at System.ServiceModel.Channels.ServiceChannel.CallOnceManager.CallOnce(TimeSpan timeout, CallOnceManager cascade)
   at System.ServiceModel.Channels.ServiceChannel.EnsureOpened(TimeSpan timeout)
   at System.ServiceModel.Channels.ServiceChannel.Call(String action, Boolean oneway, ProxyOperationRuntime operation, Object[] ins, Object[] outs, TimeSpan timeout)
   at System.ServiceModel.Channels.ServiceChannelProxy.InvokeService(IMethodCallMessage methodCall, ProxyOperationRuntime operation)
   at System.ServiceModel.Channels.ServiceChannelProxy.Invoke(IMessage message)
   at System.Runtime.Remoting.Proxies.RealProxy.HandleReturnMessage(System.Runtime.Remoting.Messaging.IMessage, System.Runtime.Remoting.Messaging.IMessage)
   at System.Runtime.Remoting.Proxies.RealProxy.PrivateInvoke(System.Runtime.Remoting.Proxies.MessageData ByRef, Int32)
   at NAPS2.Worker.IWorkerService.Init(System.String)
   at NAPS2.DI.NinjectWorkerServiceFactory.Create()
   at NAPS2.WinForms.FDesktop.RenderThumbnails()
   at System.Threading.ExecutionContext.RunInternal(System.Threading.ExecutionContext, System.Threading.ContextCallback, System.Object, Boolean)
   at System.Threading.ExecutionContext.Run(System.Threading.ExecutionContext, System.Threading.ContextCallback, System.Object, Boolean)
   at System.Threading.ExecutionContext.Run(System.Threading.ExecutionContext, System.Threading.ContextCallback, System.Object)
   at System.Threading.ThreadHelper.ThreadStart()

cyanfish, 2023-01-04:
Can you upload your [NAPS2 errorlog](https://www.naps2.com/doc/troubleshooting#error-log) here?

You can also try the latest [beta version](https://www.naps2.com/download) of NAPS2.

PeteSral, 2023-01-05:
[errorlog.txt](https://github.com/cyanfish/naps2/files/10348356/errorlog.txt)
Attached is the error log before trying the latest beta version.

PeteSral, 2023-01-05:
Beta version fixes the issue, after looking at the log I can see it was failing on checking for updates with the error `The request was aborted: Could not create SSL/TLS secure channel.`   Probably communicated on an older TLS 1.1 as my workstation does not allow under 1.2

Thanks for your help.

closed 2023-01-05 by PeteSral