cyanfish/naps2 #48: [7.0B2] Crash When Cropping Color/Image Page

AlexanderGH, 2023-01-03:
Attempting to crop a page scanned in in color (in the recovery data, these are stored as jpgs instead of pngs, i'm not sure how naps decides whether to use png or jpg) results in a crash. I wasn't able to find any relevant logs (only for other issues).

**To Reproduce**
Steps to reproduce the behavior:
1. Scan a page in color
2. Double click on the page
3. Click on crop button

**Expected behavior**
No crash.

**Screenshots**
Not applicable

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Version 22H2, 22621.963

**Additional context**
Add any other context about the problem here.

cyanfish, 2023-01-04:
As far as logs, can you check both the [NAPS2 logs](https://www.naps2.com/doc/troubleshooting#error-log) and the [Windows Event Logs](https://www.howtogeek.com/123646/htg-explains-what-the-windows-event-viewer-is-and-how-you-can-use-it/) (under the Application section)?

Does the problem recur if you save the page as a JPG, then re-import it? If so, can you upload a sample JPG here?

Are you using Windows in English or another language?

AlexanderGH, 2023-01-04:
I'm using the portable version and the naps log didn't have anything when i looked yesterday. However, the crashes do show up in the windows error viewer:

```
Faulting application name: NAPS2.exe, version: 7.0.2.0, time stamp: 0xd5751993
Faulting module name: ntdll.dll, version: 10.0.22621.900, time stamp: 0xa97a9ed6
Exception code: 0xc0000374
Fault offset: 0x000000000010c249
Faulting process id: 0x0x33A4
Faulting application start time: 0x0x1D91FBADE8B63F0
Faulting application path: C:\Users\Alexander\Apps\PortableApps\NAPS\App\NAPS2.exe
Faulting module path: C:\Windows\SYSTEM32\ntdll.dll
Report Id: fa6bd311-0e96-4b9a-9aff-b1d591436641
Faulting package full name: 
Faulting package-relative application ID: 
```

```
Fault bucket 1952135432818768641, type 4
Event Name: APPCRASH
Response: Not available
Cab Id: 0

Problem signature:
P1: NAPS2.exe
P2: 7.0.2.0
P3: d5751993
P4: StackHash_f3f9
P5: 10.0.22621.900
P6: a97a9ed6
P7: c0000374
P8: PCH_F3_FROM_ntdll+0x000000000009F954
P9: 
P10: 
```

I'm using Windows in English. I don't have any PII-free image handy, but if the above is not enough, I can go and scan in some spam i got in the mail which should hopefully repro the issue (i just tried with a random jpeg off the internet and it does not repro).

AlexanderGH, 2023-01-04:
Found some spam to span. Both the pdg and jpg seem to result in a crash hen i try try to crop. My guess is it has something to do with the high resolution, though NAPS 6.x handles it fine.

https://www.alexanderzlatin.com/files/crash1.jpg

cyanfish, 2023-01-08:
Thanks, this should be fixed in [7.0b3](https://www.naps2.com/download).

commit a24f6211 (2023-01-08): Gdi copy-on-write is not thread safe

Fixes #48
closed 2023-01-08 by cyanfish