cyanfish/naps2 #50: [7.0b2][Portable] Automatic Update Fails

AlexanderGH, 2023-01-08:
Thanks for the quick fixes for beta2, a new minor bug (or maybe WAI issue in the protable version) when updating.

**To Reproduce**
Steps to reproduce the behavior:
- Install 7b2 portable version (not the latest b3)
- Go to about box
- Check for updates
- Click on "Install 7.0b3"
- Wait for download and restart and go to the about box.
- Observe that you're still on b2

**Expected behavior**
Installs 7.0b3 and restarts.

**Screenshots**
N/A

**Desktop (please complete the following information):**
Same as #48 

**Additional context**
- Nothing in the error log.
- Update/modification timestamp of NAPS2.Portable.exe is updated so I assume that got replaced. However, nothing under App/ was modified.

cyanfish, 2023-01-08:
To double check, you're looking in the Data folder for the error log? If you try again a few times does it do the same thing? I can't reproduce the issue myself.

AlexanderGH, 2023-01-08:
So, I actually extracted 7.0.x over 6.0.x, so there's two: one under App/Data/ and one under Data/. Whenever i report a bug I check both since I'm not sure which one is new and which is old.

I tried it a few times and same issue. Also happy to try a cleanish install if you let me know which files i need to keep to hold onto my settings/configs.

However, out of curiosity, I checked the windows error log same as in #48 and i think it might actually point to the issue:
```
Application: NAPS2.Portable.exe
Framework Version: v4.0.30319
Description: The process was terminated due to an unhandled exception.
Exception Info: System.IO.IOException
   at System.IO.Directory.InternalMove(System.String, System.String, Boolean)
   at NAPS2.Portable.Program.AtomicReplaceDirectory(System.String, System.String)
   at NAPS2.Portable.Program.Main(System.String[])

Faulting application name: NAPS2.Portable.exe, version: 7.0.3.0, time stamp: 0xfd2475da
Faulting module name: KERNELBASE.dll, version: 10.0.22621.963, time stamp: 0x766ca8ae
Exception code: 0xe0434352
Fault offset: 0x00000000000906bc
Faulting process id: 0x0x400C
Faulting application start time: 0x0x1D922FFD506FB30
Faulting application path: C:\Users\Alexander\Apps\PortableApps\NAPS\NAPS2.Portable.exe
Faulting module path: C:\Windows\System32\KERNELBASE.dll
Report Id: e8399569-67b7-4b14-9b45-2f4a363bf34c
Faulting package full name: 
Faulting package-relative application ID: 
```

cyanfish, 2023-01-08:
The "Data" folder inside the "App" folder is definitely a bug, but I don't think a critical one - you should be able to ignore it for the most part.

That error is definitely related to the update failure. The most likely reason would be that a file in the App folder is in use so it can't be replaced. It's supposed to wait until NAPS2 is closed so that doesn't happen. Is there any chance you had Windows Explorer open in the App folder, or something in a text editor, etc. that could have kept something in that folder in use?

A fresh install would definitely fix it. All you'd need to copy over is config.xml and profiles.xml in the top-level Data folder.

AlexanderGH, 2023-01-08:
I was pretty sure I didn't have it open (I get hit by the "x is in use" errors a non-trivial number of times so it's one of the things i also thought about), but, as i double checked right now, and tried one more time, it did successfully update, so now I'm doubting myself.

Probably safe to close this, though I still think it might make sense to try and handle this error and at least log more details (the .net forms stuff is part of the app dir, so i assume you can't use it, but an error message with "X could not be moved, check that it is not in use by another program" would also be nice).

cyanfish, 2023-01-08:
I'll leave this open for now with a few action items:
- Fix the extra Data folder bug
- Log/display errors that happen in NAPS2.Portable.exe somehow
- Verify that NAPS2 worker processes are guaranteed to be killed before the update process is attempted (they're supposed to be, but there's a chance of a bug with this being why the files were in use)

gitname, 2023-01-13:
I am observing this on my system (Windows 10) also.

**Screenshot:**

[image]

**Error Log:**

I checked `{my-portable-apps}/naps2-7.0b2-win/Data/errorlog.txt` and saw the following log entry (I applied `py` syntax highlighting here because—although it isn't Python code—out of the syntaxes I tried, `py` colors the text in the way I found most helpful):

```py
2023-01-13 08:44:20.0269 4628 Update error System.Net.WebException: The underlying connection was closed: An unexpected error occurred on a receive. ---> System.ComponentModel.Win32Exception: The client and server cannot communicate, because they do not possess a common algorithm
   at System.Net.SSPIWrapper.AcquireCredentialsHandle(SSPIInterface SecModule, String package, CredentialUse intent, SecureCredential scc)
   at System.Net.Security.SecureChannel.AcquireCredentialsHandle(CredentialUse credUsage, SecureCredential& secureCredential)
   at System.Net.Security.SecureChannel.AcquireCredentialsHandle(CredentialUse credUsage, X509Certificate2 selectedCert, Flags flags)
   at System.Net.Security.SecureChannel.AcquireClientCredentials(Byte[]& thumbPrint)
   at System.Net.Security.SecureChannel.GenerateToken(Byte[] input, Int32 offset, Int32 count, Byte[]& output)
   at System.Net.Security.SecureChannel.NextMessage(Byte[] incoming, Int32 offset, Int32 count)
   at System.Net.Security.SslState.StartSendBlob(Byte[] incoming, Int32 count, AsyncProtocolRequest asyncRequest, Boolean renegotiation)
   at System.Net.Security.SslState.ForceAuthentication(Boolean receiveFirst, Byte[] buffer, AsyncProtocolRequest asyncRequest, Boolean renegotiation)
   at System.Net.Security.SslState.ProcessAuthentication(LazyAsyncResult lazyResult)
   at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   at System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   at System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state)
   at System.Net.TlsStream.ProcessAuthentication(LazyAsyncResult result)
   at System.Net.TlsStream.BeginWrite(Byte[] buffer, Int32 offset, Int32 size, AsyncCallback asyncCallback, Object asyncState)
   at System.Net.ConnectStream.WriteHeaders(Boolean async)
   --- End of inner exception stack trace ---
   at System.Net.HttpWebRequest.EndGetResponse(IAsyncResult asyncResult)
   at System.Net.WebClient.GetWebResponse(WebRequest request, IAsyncResult result)
   at System.Net.WebClient.DownloadBitsResponseCallback(IAsyncResult result)
   at NAPS2.Update.UpdateOperation.DownloadCompleted(Object sender, AsyncCompletedEventArgs e)
```

**Environment:**

- **NAPS2 version**: `7.0.2.0`
- **OS**: Windows 10 Enterprise LTSC Evaluation (Version `10.0.17763` Build `17763`)
- **Other**: NAPS2 version `6.1.2.25834` is also installed on the system, but was not running at the time of the error

**Footnotes:** I have been using NAPS2 for several years and it is my favorite scanning program. I like the (IMO) straightforward UI and the absence of "bloat." Thank you for creating it and allowing people to use it for free!

AlexanderGH, 2023-01-13:
I think yours is likely different in root cause than my issue, though the outcome is the same. Yours seems to be a TLS negotiation failure (super weird unless the server is misconfigured though, or your .NET is _really_ old, where i doubt NAPS would even run).

However, yours does show an error message, which means it should also be possible to show a similar error message (i would hope) when the failure I saw triggers.

cyanfish, 2023-02-06:
I believe all the issues mentioned here should be addressed in [7.0b4](https://www.naps2.com/download) (though I suppose some issues will only be testable once the next version is out).

closed 2023-02-06 by cyanfish