cyanfish/naps2 #51: Official Microsoft Store Version?

AlexanderGH, 2023-01-08:
Unfortunately (though it's WAI with open source), a lot of great open source apps end up being exploited by people to make a quick buck by rebranding them and then selling them. In the more extreme cases, those apps end up being adware, or malware. This is the case for apps like 7zip, and from what I can tell, NAPS, at least based on the screenshot (not going to install it to confirm), ripped straight from your site ttps://www.microsoft.com/store/productId/9PL6FK27J78G (apologies if this app is actually you).

**Describe the solution you'd like**
Similar to the approach paint.net took, provide an official paid version in the microsoft store. It provides more exposure, more users, and also some spare change. It also discourages malicious versions from being uploaded.

**Describe alternatives you've considered**
N/A

**Additional context**
- I'm not super familiar with the Microsoft Store, but it seems similar to the Play store where it's only a cheap one-time fee instead of Apple's recurring high fee.
- I assume if IARC allowed non-commertial development of NAPS, they're also be fine with commercial.
- The default windows 11 scan app is not that great and users know it (3 stars, vs 4.1 for the one i linked).

cyanfish, 2023-01-08:
Thanks for bringing that to my attention. I'm planning on providing an official Microsoft Store app with the next non-beta version (and equivalents for Mac/Linux as well), I'm not sure yet if I'll have that be free or paid. I don't think IARC should be an issue.

riccardoquercetti, 2023-07-20:
Did you known that your app is already present in the Microsoft app store rebranded under the name of "Any Scan"?

[image]

alleluia20, 2024-08-11:
@cyanfish: we can now find the NAPS2 app at the Windows store. It is supplied by "Vanced apps". Is it reliable?

On the one hand, they keep the name NAPS2 and they link to the official website for the privacy policy. On the other hand, the app has very few reviews and the supplier looks obscure.

If you think it is reliable, I would install the app from the Windows store, in order to automatically get the updates. But, in doubt, I will continue downloading it from the official website: @cyanfish: **do you think we can trust this app?**

Thank you, and regards.

cyanfish, 2024-08-11:
@alleluia20 It's not official. I have no indication they're doing anything nefarious but there's no guarantee it will keep up to date with the latest NAPS2 version.

kashmirix, 2024-11-17:
It's not that the counterfeit app _is_ nefarious at the moment. It's more about its potential to destroy the NAPS2 brand altogether. It's not that such things never happened:

[https://github.com/greatsuspender/thegreatsuspender/issues/1263](https://github.com/greatsuspender/thegreatsuspender/issues/1263)

I'm not an expert, but while the code is distributed under GPL, some sort of trademarking the "NAPS2" name (as a wordmark) may make sense. Trademarking is expressly permitted by GPL and would offer some protections against redistribution under the same name whenever the developer feels like enforcing it.

cyanfish, 2025-06-03:
NAPS2 is now (finally!) officially available in the Microsoft Store. [Download here](https://apps.microsoft.com/detail/9n3qq9w0b23q?hl=en-US&gl=US).

I'm charging 10 USD for it, similar to the Paint.NET model where you can download in the store as an alternative to donating. If you've already donated feel free to send an email to the address in the donation receipt and I can give you a one-time code for a free download.

Let me know if you have any issues, and I would appreciate positive ratings/reviews on the app since it doesn't have any yet.

cyanfish, 2025-06-03:
> Great! May make sense to specify the full list of supported languages in product specs, else Store might not show it in search results for non-US users.

Good point, I thought I had that set up but evidently not. Created #632 to track.