cyanfish/naps2 #57: 7.0.3 Console working with Wia profile but not Twain profile

mrbastian, 2023-01-24:
First let me say that this project is absolutely amazing and I'm excited to see the updates coming with v7. 

I'm running 7.0.3 64-bit with 2 profiles named 1) BW300 using twain (Paperstream), and 2) WIA using WIA

The console hangs when trying to scan using the twain profile. Please let me know if there is any additional troubleshooting steps I can do.

In the GUI, I am able to scan with no issues using either profile.

Using the console with WIA profile/driver, with the following command line:
NAPS2.Console.exe --profile WIA --output c:\users\mbastian\downloads\x.tif --force --progress --verbose

Output is :
Beginning scan...
Starting scan 1 of 1...
Scanned page 1.
1 page(s) scanned.
Exporting...
Exporting image 1 of 1...
Finished saving images to c:\users\mbastian\downloads\x.tif

Using the console with WIA profile/driver, with the following command line:
NAPS2.Console.exe --profile BW300 --output c:\users\mbastian\downloads\x.tif --force --progress --verbose

Output is:
Beginning scan...
Starting scan 1 of 1...    

and nothing after that. The process hangs until I kill it.

Desktop OS: Windows 11

mrbastian, 2023-01-26:
I managed to build the project locally and run it in debug.

The problem seems to be related to starting the Native worker process in NAPS2.Remoting.Worker.WorkerFactory.StartWorkerProcess. It tries starting AssemblyHelper.EntryFile which is NAPS2.Console.dll with the arguments $"worker {parentId}". But that throws an exception from the output "At least one of the -o/--output, -e/--email, or -a/--autosave options must be specified." when it was expecting "ready" or "error".  The WinX86 worker meanwhile does start and return "ready".

I see that NAPS2.Console.dll doesn't have a command line argument for "worker", but haven't tried to dig in any further.

cyanfish, 2023-01-29:
I don't think that error is related to the problem, though I will fix it in the next beta. For testing you can comment out [this line](https://github.com/cyanfish/naps2/blob/df03d6470fba647e9f1b2ea05ff4b75937901113/NAPS2.Sdk/Remoting/Worker/WorkerFactory.cs#L144) and the error should go away (only the WinX86 worker is needed).

When you run the program in debug, it should write extra TWAIN logging to the [error log](https://www.naps2.com/doc/troubleshooting#error-log), if you can have a look and copy that here.

One thing also worth trying is with "Use native UI" on/off in the profile - see if that works.

Assuming you had "Use native UI" unchecked, I'm surprised there would be any difference between GUI and Console, as in both cases the TWAIN scan happens in a worker process and off the top of my head there should be [no difference](https://github.com/cyanfish/naps2/blob/df03d6470fba647e9f1b2ea05ff4b75937901113/NAPS2.Lib.WinForms/WinForms/WinFormsTwainHandleManager.cs#L32) in the TWAIN state...

Also keep in mind, debugging TWAIN can be tricky as the debugger won't attach to the worker process by default. Though hopefully the TWAIN logging gives some hints.

cyanfish, 2023-01-29:
Also, one more thing you can try is selecting different options under Advanced -> Twain Implementation in the profile settings, and see if any of the other options work.

mrbastian, 2023-01-29:
Another thing I tried was running Naps2.App.WinForms to see what that did (since the twain profile worked fine from the UI). In debug, I saw the WorkerFactory was using Naps2.exe when workerType == Native. So for the fun of it, I changed the logic so that when running the Console / command line, it used Naps2.exe for Native as well. That scanned with the Twain profile with no issues.

When I commented out the line you suggested, that also hung like before. In profile settings, I was already using "use predefined settings", but I tried "Use native UI" as well with the same results.

The LAST thing I tried which had successfult results and I think _might_ be the correct fix was to add the following lines here: 

https://github.com/cyanfish/naps2/blob/df03d6470fba647e9f1b2ea05ff4b75937901113/NAPS2.Lib/EntryPoints/ConsoleEntryPoint.cs#L16

        if (args.Length > 0 && args[0] == "worker")
        {
            return WorkerEntryPoint.Run(args.Skip(1).ToArray(), imageModule);
        }

cyanfish, 2023-01-29:
Did you by any chance select "x64" under Twain Implementation in your profile settings?

mrbastian, 2023-01-29:
Yes, it was set to x64. When I first ran the beta, no twain drivers even came up like they did before so I found the 64-bit paperstream and installed that. Then I was able to select the device. So both 32-bit and 64-bit drivers should be installed now. Is there something else you would like me to try?

mrbastian, 2023-01-29:
If I change it to "default", then it DOES work without my change.

cyanfish, 2023-01-29:
That makes sense, that's the one scenario where it will use a native worker. I'll fix the worker path issue in the next beta.

closed 2023-02-06 by cyanfish

cyanfish, 2023-02-06:
This should be fixed in [7.0b4](https://www.naps2.com/download), please try and see if it works.

mrbastian, 2023-02-07:
Tested and confirmed the reported issue is fixed in 7.0b4. Thanks!

Another thing I noticed that I should have brought before - the --progress option "Display a graphical window for scanning progress." does not appear to do anything. Should I add that as a separate issue, or is there anything you'd like me to try first?
My command line is the same as before:
NAPS2.Console.exe --profile BW300 --output c:\users\mbastian\downloads\x.tif --force --progress --verbose

cyanfish, 2023-02-07:
Please create a new issue, I think that was an oversight on my part.

commit bacd0045 (2023-02-06): Fix worker paths for console

Fixes #57