cyanfish/naps2 #58: [7.0.3.0] Framework exception (DragDrop-Registration)

akuropka, 2023-01-28:
Upon scanning an image the following exception is thrown by .NET framework:
> Informationen über das Aufrufen von JIT-Debuggen
> anstelle dieses Dialogfelds finden Sie am Ende dieser Meldung.
> 
> ************** Ausnahmetext **************
> System.InvalidOperationException: Fehler bei der DragDrop-Registrierung ---> System.Threading.ThreadStateException: Für den aktuellen Thread muss der STA-Modus (Single Thread Apartment) festgelegt werden, bevor OLE-Aufrufe ausgeführt werden können. Stellen Sie sicher, dass die Hauptfunktion mit STAThreadAttribute gekennzeichnet ist.
>    bei System.Windows.Forms.Control.SetAcceptDrops(Boolean accept)
>    --- Ende der internen Ausnahmestapelüberwachung ---
>    bei System.Windows.Forms.Control.SetAcceptDrops(Boolean accept)
>    bei System.Windows.Forms.Control.OnHandleCreated(EventArgs e)
>    bei System.Windows.Forms.Form.OnHandleCreated(EventArgs e)
>    bei System.Windows.Forms.Control.WmCreate(Message& m)
>    bei System.Windows.Forms.Control.WndProc(Message& m)
>    bei System.Windows.Forms.Form.WmCreate(Message& m)
>    bei Eto.WinForms.Forms.EtoFormBase.WndProcImpl(Message& m)
>    bei System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)
> 
> 
> ************** Geladene Assemblys **************
> mscorlib
>     Assembly-Version: 4.0.0.0.
>     Win32-Version: 4.8.9105.0 built by: NET481REL1LAST_C.
>     CodeBase: file:///C:/Windows/Microsoft.NET/Framework64/v4.0.30319/mscorlib.dll.
> ----------------------------------------
> NAPS2
>     Assembly-Version: 7.0.3.0.
>     Win32-Version: 7.0.3.0.
>     CodeBase: file:///C:/Program%20Files/NAPS2/NAPS2.exe.
> ----------------------------------------
> NAPS2.Lib.WinForms
>     Assembly-Version: 7.0.3.0.
>     Win32-Version: 7.0.3.0.
>     CodeBase: file:///C:/Program%20Files/NAPS2/lib/NAPS2.Lib.WinForms.DLL.
> ----------------------------------------
> Eto
>     Assembly-Version: 2.7.0.0.
>     Win32-Version: 2.7.8333.40430.
>     CodeBase: file:///C:/Program%20Files/NAPS2/lib/Eto.DLL.
> ----------------------------------------
> netstandard
>     Assembly-Version: 2.0.0.0.
>     Win32-Version: 4.8.9032.0.
>     CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/netstandard/v4.0_2.0.0.0__cc7b13ffcd2ddd51/netstandard.dll.
> ----------------------------------------
> NAPS2.Lib
>     Assembly-Version: 7.0.3.0.
>     Win32-Version: 7.0.3.0.
>     CodeBase: file:///C:/Program%20Files/NAPS2/lib/NAPS2.Lib.DLL.
> ----------------------------------------
> System
>     Assembly-Version: 4.0.0.0.
>     Win32-Version: 4.8.9065.0 built by: NET481REL1LAST_C.
>     CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System/v4.0_4.0.0.0__b77a5c561934e089/System.dll.
> ----------------------------------------
> Autofac
>     Assembly-Version: 6.4.0.0.
>     Win32-Version: 6.4.0.0.
>     CodeBase: file:///C:/Program%20Files/NAPS2/lib/Autofac.DLL.
> ----------------------------------------
> Microsoft.Bcl.AsyncInterfaces
>     Assembly-Version: 6.0.0.0.
>     Win32-Version: 6.0.21.52210.
>     CodeBase: file:///C:/Program%20Files/NAPS2/lib/Microsoft.Bcl.AsyncInterfaces.DLL.
> ----------------------------------------
> System.Threading.Tasks.Extensions
>     Assembly-Version: 4.2.0.1.
>     Win32-Version: 4.6.28619.01.
>     CodeBase: file:///C:/Program%20Files/NAPS2/lib/System.Threading.Tasks.Extensions.DLL.
> ----------------------------------------
> NAPS2.Sdk
[... 159 lines cut ...]
> System.Runtime.CompilerServices.Unsafe
>     Assembly-Version: 6.0.0.0.
>     Win32-Version: 6.0.21.52210.
>     CodeBase: file:///C:/Program%20Files/NAPS2/lib/System.Runtime.CompilerServices.Unsafe.DLL.
> ----------------------------------------
> 
> ************** JIT-Debuggen **************
> Um das JIT-Debuggen (Just-In-Time) zu aktivieren, muss in der
> Konfigurationsdatei der Anwendung oder des Computers
> (machine.config) der jitDebugging-Wert im Abschnitt system.windows.forms festgelegt werden.
> Die Anwendung muss mit aktiviertem Debuggen kompiliert werden.
> 
> Zum Beispiel:
> 
> <configuration>
>     <system.windows.forms jitDebugging="true" />
> </configuration>
> 
> Wenn das JIT-Debuggen aktiviert ist, werden alle nicht behandelten
> Ausnahmen an den JIT-Debugger gesendet, der auf dem
> Computer registriert ist, und nicht in diesem Dialogfeld behandelt.
> 

Screencast:
https://user-images.githubusercontent.com/22570574/215283994-802ee798-bb14-4ef9-9147-562e3db6f2ed.mp4

cyanfish, 2023-01-29:
Does this work normally with version 6.1.2?

akuropka, 2023-01-29:
Yes.

cyanfish, 2023-02-04:
Can you try this test version and see if there's any difference? [naps2-7.0b3-testtwain.zip](https://github.com/cyanfish/naps2/files/10609191/naps2-7.0b3-testtwain.zip)

akuropka, 2023-02-05:
Hello @cyanfish,
the issue persists. I tried a little bit around and I can break it down to the AutoSave festure which causes the bug.
If I choose to proceed at the error message and select then processing in background the task seems to get completed as expected. An errorlog file is not created.
I hope this helps. Thank you for your support.

closed 2023-02-06 by cyanfish

cyanfish, 2023-02-06:
This should be fixed in [7.0b4](https://www.naps2.com/download), please try and see if it works.

akuropka, 2023-02-06:
perfect, thanks.