cyanfish/naps2 #59: Scanning unter MacOS Monterey (12) doesn't work

WSHub43, 2023-01-29:
I am using MacOS Monterey on a Macbook Pro M1. I have a Canon MF657Cdw and use Naps 7.0b3. I have installed the last Canon printer and scanner packages and printing and scanning with the Canon apps works fine.

Installation of NAP2 and adding the printer as a scanner works fine (used Apple as engine, the second option doesn't detect the printer). But if I want so scan something just the dialog windows comes without a progress (progressbar stucks at the start). The scanner unit doesn't do anything. First I thought the printer is to new for NAPS2, so I tried it on a Windows 11 machine (v7b3 as well) and here everything is fine (with WIA).

Did I make any mistake?

cyanfish, 2023-01-29:
Does it work with the basic Apple scanning software (Systems Settings -> Printers and Scanners -> Open Scanner)?

Are you scanning with the flatbed or feeder - and does the other have the same problem?

Can you see if there's an errorlog.txt file and upload it here? You can find it in the `/Users/yourusername/.config/naps2` folder (in Finder you may need to press Cmd+Shift+. to see the .config folder).

WSHub43, 2023-01-29:
Yes it does (both flatbed and feeder), and yes with NAPS2 flatbed and feeder don't work (same behaviour).

Here is the error text:

2023-01-27 14:28:49.8376 21969 Marshalling managed exception NAPS2.Scan.Exceptions.NoDevicesFoundException: No scanning device was found.
   at NAPS2.Scan.ScanPerformer.PromptForDevice(ScanOptions options)
   at NAPS2.Scan.ScanPerformer.PopulateDevice(ScanProfile scanProfile, ScanOptions options)
   at NAPS2.Scan.ScanPerformer.PerformScan(ScanProfile scanProfile, ScanParams scanParams, IntPtr dialogParent, CancellationToken cancelToken)+MoveNext()
   at NAPS2.Scan.ScanPerformer.PerformScan(ScanProfile scanProfile, ScanParams scanParams, IntPtr dialogParent, CancellationToken cancelToken)+System.Threading.Tasks.Sources.IValueTaskSource<System.Boolean>.GetResult()
   at NAPS2.EtoForms.Ui.ProfilesForm.DoScan()
   at NAPS2.EtoForms.Ui.ProfilesForm.DoScan()
   at System.Threading.Tasks.Task.<>c.<ThrowAsync>b__128_0(Object state)
   at Foundation.NSAsyncSynchronizationContextDispatcher.Apply()
   at InvokeStub_NSAsyncSynchronizationContextDispatcher.Apply(Object, Object, IntPtr*)
   at System.Reflection.MethodInvoker.Invoke(Object obj, IntPtr* args, BindingFlags invokeAttr)
--- End of stack trace from previous location ---
   at ObjCRuntime.Runtime.InvokeMethod(MethodBase method, Object instance, IntPtr native_parameters)
   at ObjCRuntime.Runtime.InvokeMethod(MonoObject* methodobj, MonoObject* instanceobj, IntPtr native_parameters)
   at ObjCRuntime.Runtime.bridge_runtime_invoke_method(MonoObject* method, MonoObject* instance, IntPtr parameters, IntPtr& exception_gchandle)
2023-01-27 14:28:49.8547 21969 Marshalling ObjC exception: No scanning device was found. (NAPS2.Scan.Exceptions.NoDevicesFoundException)
   at NAPS2.Scan.ScanPerformer.PromptForDevice(ScanOptions options)
   at NAPS2.Scan.ScanPerformer.PopulateDevice(ScanProfile scanProfile, ScanOptions options)
   at NAPS2.Scan.ScanPerformer.PerformScan(ScanProfile scanProfile, ScanParams scanParams, IntPtr dialogParent, CancellationToken cancelToken)+MoveNext()
   at NAPS2.Scan.ScanPerformer.PerformScan(ScanProfile scanProfile, ScanParams scanParams, IntPtr dialogParent, CancellationToken cancelToken)+System.Threading.Tasks.Sources.IValueTaskSource<System.Boolean>.GetResult()
   at NAPS2.EtoForms.Ui.ProfilesForm.DoScan()
   at NAPS2.EtoForms.Ui.ProfilesForm.DoScan()
   at System.Threading.Tasks.Task.<>c.<ThrowAsync>b__128_0(Object state)
   at Foundation.NSAsyncSynchronizationContextDispatcher.Apply()
   at InvokeStub_NSAsyncSynchronizationContextDispatcher.Apply(Object, Object, IntPtr*)
   at System.Reflection.MethodInvoker.Invoke(Object obj, IntPtr* args, BindingFlags invokeAttr)
--- End of stack trace from previous location ---
   at ObjCRuntime.Runtime.InvokeMethod(MethodBase method, Object instance, IntPtr native_parameters)
   at ObjCRuntime.Runtime.InvokeMethod(MonoObject* methodobj, MonoObject* instanceobj, IntPtr native_parameters)
   at ObjCRuntime.Runtime.bridge_runtime_invoke_method(MonoObject* method, MonoObject* instance, IntPtr parameters, IntPtr& exception_gchandle)
 
2023-01-27 14:28:49.8547 21969 Marshalling managed exception NAPS2.Scan.Exceptions.NoDevicesFoundException: No scanning device was found.
   at NAPS2.Scan.ScanPerformer.PromptForDevice(ScanOptions options)
   at NAPS2.Scan.ScanPerformer.PopulateDevice(ScanProfile scanProfile, ScanOptions options)
   at NAPS2.Scan.ScanPerformer.PerformScan(ScanProfile scanProfile, ScanParams scanParams, IntPtr dialogParent, CancellationToken cancelToken)+MoveNext()
   at NAPS2.Scan.ScanPerformer.PerformScan(ScanProfile scanProfile, ScanParams scanParams, IntPtr dialogParent, CancellationToken cancelToken)+System.Threading.Tasks.Sources.IValueTaskSource<System.Boolean>.GetResult()
   at NAPS2.EtoForms.Ui.ProfilesForm.DoScan()
   at NAPS2.EtoForms.Ui.ProfilesForm.DoScan()
   at System.Threading.Tasks.Task.<>c.<ThrowAsync>b__128_0(Object state)
   at Foundation.NSAsyncSynchronizationContextDispatcher.Apply()
   at InvokeStub_NSAsyncSynchronizationContextDispatcher.Apply(Object, Object, IntPtr*)
   at System.Reflection.MethodInvoker.Invoke(Object obj, IntPtr* args, BindingFlags invokeAttr)
--- End of stack trace from previous location ---
   at ObjCRuntime.Runtime.InvokeMethod(MethodBase method, Object instance, IntPtr native_parameters)
   at ObjCRuntime.Runtime.InvokeMethod(MonoObject* methodobj, MonoObject* instanceobj, IntPtr native_parameters)
   at ObjCRuntime.Runtime.bridge_runtime_invoke_method(MonoObject* method, MonoObject* instance, IntPtr parameters, IntPtr& exception_gchandle)
   at ObjCRuntime.Messaging.IntPtr_objc_msgSend_IntPtr(IntPtr receiver, IntPtr selector, IntPtr arg1)
   at AppKit.NSApplication.RunModalSession(IntPtr session)
   at Eto.Mac.Forms.ModalEventArgs.RunSessionInternal()
   at Eto.Mac.Forms.ModalEventArgs.RunSession()
   at Eto.Mac.Forms.ModalEventArgs.RunModal()
   at Eto.Mac.Forms.MacModal.Run(Window window, NSWindow nativeWindow, ModalEventArgs& helper, Boolean isSheet)
   at Eto.Mac.Forms.DialogHandler.ShowModal()
   at Eto.Forms.Dialog.ShowModal()
   at NAPS2.EtoForms.Desktop.DesktopSubFormController.ShowProfilesForm()
   at NAPS2.EtoForms.ActionCommand.<>c__DisplayClass1_0.<.ctor>b__0(Object sender, EventArgs args)
   at Eto.Forms.Command.OnExecuted(EventArgs e)
   at Eto.Forms.Command.Execute()
   at NAPS2.Util.MacExtensions.ActionTarget.Action()
[... cut ...]
   at System.Reflection.MethodInvoker.Invoke(Object obj, IntPtr* args, BindingFlags invokeAttr)
--- End of stack trace from previous location ---
   at ObjCRuntime.Runtime.InvokeMethod(MethodBase method, Object instance, IntPtr native_parameters)
   at ObjCRuntime.Runtime.InvokeMethod(MonoObject* methodobj, MonoObject* instanceobj, IntPtr native_parameters)
   at ObjCRuntime.Runtime.bridge_runtime_invoke_method(MonoObject* method, MonoObject* instance, IntPtr parameters, IntPtr& exception_gchandle)

WSHub43, 2023-02-07:
Update from me: I installed the newest Beta 4 and on my Intel Mac Mini Duplex-scanning works fine. So I thought the installed Canon software on my Macbook Pro M1 is the issue. I deinstalled everything (also in Library-Folders), as well as NAPS2 and just added the printer via the printer window in MacOS. Didn't install anything from Canon. But the issue is the same. Is it possible because of M1?

cyanfish, 2023-02-07:
Interesting, it could be some interaction with M1 and the Canon software. Can you try installing the Intel version of NAPS2 and see if it's any different? (It should work on M1 through Rosetta, just might be slower etc)

WSHub43, 2023-02-08:
Good news: I found the reason.

An update from Monterey to Ventura did the trick. Yesterday I tested on another Mac Mini (Intel and Ventura), where I installed all Canon software (like I did von my Macbook Pro) and it worked. So I wasn't sure if it has something to do with M1 or Ventura.

On my Macbook Pro M1 (Ventura) I have installed the intel version of NAPS2, but I guess the arm one will work as well.

From beta 3 to beta 4 the duplex-function started working.

closed 2023-03-06 by cyanfish