cyanfish/naps2 #60: Image actions not working on Linux Mint

srdlgh, 2023-01-29:
Running 7.0b3 on Linux Mint 21.1, Image submenu actions between separators don't activate (i.e, crop, brightness, hue, b&w, sharpen, correction). Others do (i.e, view, reset).  Same choices don't activate in View pane.

Assuming this can be fixed -- or it's some error of mine that someone can point out -- I'll be so pleased to finally be able to use this on Mint -- I never was able to get the portable version running with Mono. Thanks for all the work so far.

I had also hoped there would be keyboard shortcuts for all the actions -- but with some amount of scripting, it does play nicely with Autokey_X11 on Mint,  and presumably with Autokey (the Python version) and other similar tools. Now I can press one key to clear, scan, threshold and save while I pay attention elsewhere.

cyanfish, 2023-01-29:
Thanks, this will be fixed in the next beta. I've created https://github.com/cyanfish/naps2/issues/61 to track keyboard shortcut improvements.

As far as automation, Auto Save (in the profile) does almost everything you mention, although for thresholding (Black & White) there isn't yet a way to choose a custom threshold value in the profile.

srdlgh, 2023-01-29:
Ben,

Thanks for the speedy reply -- and of course for all your work on NAPS2.
I'm looking forward towards making it my primary scanning program.

I guess I could have picked a better macro example that presently doesn't
seem to have a built-in equivalent  (e.g, select page 1, save selected and
delete page 1).  The autosave function does seem it would work for most
cases I had mentioned, but I wanted a wider range of possibilities and more
flexibility. That is not in any way a criticism of NAPS2 --  I generally
have done macros or scripts for most programs I use regularly.

I realize that keyboard shortcuts aren't a given for everyone -- but for
those of us who started in computing before the Macintosh or came up
through Unix/Linux, they seem a natural part of a well-written effort.  If
you do get around to them, good -- but personally I'm happy using my
keyboard macros    while you attend to more important priorities.

--Steve R

antze-k, 2023-02-02:
As a person who also encountered this bug and did some research, I'd like to let you know that the problem is in NAPS2.Lib/EtoForms/Ui/ImageFormBase.cs, OnLoad() method. Screen.WorkingArea.Width (and Height) seems to refer to Windows Forms value, if I replace it with Gdk.Screen.Default.Width (adding GtkSharp thus breaking some abstractions of course), everything seems to be working fine.

Probably you found this already, but anyway I hope this can be helpful.

cyanfish, 2023-02-02:
Yep, that's the fix I used :) Specifically `Screen ?? Screen.PrimaryScreen`

closed 2023-02-06 by cyanfish

cyanfish, 2023-02-06:
This should be fixed in [7.0b4](https://www.naps2.com/download), please try and see if it works.

srdlgh, 2023-02-06:
So far, looks good.  Thanks for the quick fix.

I don't know if it is a design decision or perhaps a lower priority, but the crop window doesn't show the sliders that I see on the Windows edition. Certainly not a deal-breaker, though.

cyanfish, 2023-02-06:
The crop window should be the same on Windows and Linux if you're looking at beta 7 (it was changed from version 6), both have the black handles on the outside of the image.

commit e7b099c0 (2023-02-06): Gtk: Fix image forms

Fixes #60