cyanfish/naps2 #67: Duplex scanning not working since 6.0b3 till now

cz-vilda, 2023-01-31:
**Describe the bug**
Unable to scan in duplex mode. Only first page is scanned. Last version able to scan in duplex mode is 6.0b3.

**Expected behavior**
Scan both of pages.

**Desktop (please complete the following information):**
 - OS: Windows
 - Version: 11 Pro
 - scanner Fujitsu fi-6130

**Additional context**
See error from log files.

**Version 6.0b4 and above:**
```
2023-01-31 22:07:46.4295 Error setting property System.ArgumentException: Could not set property 3096 (Pages) value to "0"
Parameter name: value
   v NAPS2.Scan.Wia.Native.WiaProperty.set_Value(Object value)
   v NAPS2.Scan.Wia.Native.WiaExtensions.SetProperty(WiaItemBase item, Int32 propId, Int32 value)
```
   
**Version 7.x:**
  ```
 2023-01-31 21:51:36.2387 2000 Error setting property System.ArgumentException: Could not set property 3096 (Pages) value to "0"
Parameter name: value
   v NAPS2.Wia.WiaProperty.set_Value(Object value)
   v NAPS2.Wia.WiaExtensions.SetProperty(WiaItemBase item, Int32 propId, Int32 value)
   v NAPS2.Scan.Internal.Wia.WiaExtensions.SafeSetProperty(WiaItemBase item, Int32 propId, Int32 value)
```

cyanfish, 2023-02-03:
Can you try with this test version? [naps2-7.0b3-testwia.zip](https://github.com/cyanfish/naps2/files/10575463/naps2-7.0b3-testwia.zip)

And then, whether or not it works, can you take the errorlog.txt file from the Data folder and attach it here?

cz-vilda, 2023-02-10:
Still not working.
**errorlog.txt:**
`2023-02-10 22:30:47.9134 11044 IPS_PAGES: min 0, max 50, step 1, nom 1, flags ReadWrite, Range 
2023-02-10 22:30:47.9915 11044 Could not set property 3096 (Pages) value to "50"
Název parametru: value System.ArgumentException: Could not set property 3096 (Pages) value to "50"
Název parametru: value
   v NAPS2.Wia.WiaProperty.set_Value(Object value)
   v NAPS2.Scan.Internal.Wia.WiaScanDriver.WiaScanContext.ConfigureProps(WiaDevice device, WiaItem item)
   v NAPS2.Scan.Internal.Wia.WiaScanDriver.WiaScanContext.DoTransfer(WiaDevice device, WiaItem item)
   v NAPS2.Scan.Internal.Wia.WiaScanDriver.WiaScanContext.<Scan>d__6.MoveNext()
--- Konec trasování zásobníku z předchozího místa, ze kterého byla vyvolána výjimka ---
   v System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   v System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   v NAPS2.Scan.Internal.Wia.WiaScanDriver.<>c__DisplayClass3_0.<<Scan>b__0>d.MoveNext()
--- Konec trasování zásobníku z předchozího místa, ze kterého byla vyvolána výjimka ---
   v System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   v System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   v NAPS2.Scan.Internal.RemoteScanController.<Scan>d__5.MoveNext()
--- Konec trasování zásobníku z předchozího místa, ze kterého byla vyvolána výjimka ---
   v System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   v System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   v NAPS2.Scan.ScanController.<>c__DisplayClass10_0.<<Scan>b__6>d.MoveNext()
`

cyanfish, 2023-02-10:
Thanks, can you try selecting "1.0" for "Wia Version" in the Advanced profile settings window? Doesn't matter if it's the test version or the normal NAPS2 installation. If that fails too I'd like to see the error log.

cz-vilda, 2023-02-16:
Stil not working...
`2023-02-16 20:15:22.9018 6636 Došlo k chybě ovladače skeneru. NAPS2.Wia.WiaException: WIA error code 80070057
   v NAPS2.Wia.WiaTransfer.Download()
   v NAPS2.Scan.Internal.Wia.WiaScanDriver.WiaScanContext.DoTransfer(WiaDevice device, WiaItem item)
   v NAPS2.Scan.Internal.Wia.WiaScanDriver.WiaScanContext.<Scan>d__6.MoveNext()
--- Konec trasování zásobníku z předchozího místa, ze kterého byla vyvolána výjimka ---
   v System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   v System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   v NAPS2.Scan.Internal.Wia.WiaScanDriver.<>c__DisplayClass3_0.<<Scan>b__0>d.MoveNext()
`

cyanfish, 2023-04-15:
Hi, if you're still interested in trying to fix this, I'm circling back to this as I got my hands on a fi-6130Z, though I can't reproduce the issue. What driver version do you have installed? You can find it in [Device Manager](https://support.microsoft.com/en-us/windows/open-device-manager-a7f2db46-faaf-24f0-8b7b-9e4a6032fc8c).
[image]

cz-vilda, 2025-03-31:
Hello, after 2 years Im back with this issue. Are you still willing to fix it?
Issue with 8.1.2 is still same.
OS: Win11
Driver: WIA, FUJISTU, date 2012-09-03, version 9.24.7 (installed automatically)

When scanning "use native UI" it works well (all pages and all sides eg. 3 papers, 6 scanned sides), when "use preferred setting", it scans only one page and one side.

Error log:
`2025-03-31 21:27:55.7687 3320 Error setting property 3096 System.ArgumentException: Value does not fall within the expected range.
   at void NAPS2.Wia.Native.IWiaPropertyStorage.WriteMultiple(uint cpspec, PROPSPEC[] rgpspec, PROPVARIANT[] rgpropvar, uint propidNameFirst)
   at uint NAPS2.Wia.Native.NativeWiaMethods.SetPropertyInt(IntPtr propStoragePtr, int propId, int value)
   at void NAPS2.Wia.WiaProperty.set_Value(object value)
   at void NAPS2.Wia.WiaExtensions.SetProperty(WiaItemBase item, int propId, int value)
   at void NAPS2.Scan.Internal.Wia.WiaScanDriver+WiaScanContext.SafeSetProperty(WiaItemBase item, int propId, int value)
`

closed 2026-01-13 by cz-vilda