cyanfish/naps2 #71: PDF page size doesn't match scanned image size

AltoRetrato, 2023-02-03:
**Describe the bug**
I created a profile with an A4 page size (210 x 297 mm). After scanning images with that profile and saving them as a PDF, the PDF page size is 656.0 x 927.8 mm (25.83 x 36.53 inches), instead of being 210 x 297 mm.

**To Reproduce**
Steps to reproduce the behavior:
1. Create or edit a profile
2. Click on 'Page size', select "A4 (210x297 mm)"
4. Scan one or more images
5. Click 'Save PDF'
6. Open the PDF and see the page size

**Expected behavior**
A PDF file with A4 pages, measuring 210 x 297 mm.

**Screenshots**
[image]

[image]

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Version: both in 6.1.2.25834 and 7.0b3

cyanfish, 2023-02-03:
Try selecting "Stretch to page size" in the Advanced profile window.

AltoRetrato, 2023-02-05:
Solved! Thanks! :)

But why isn't this setting enabled by default?

cyanfish, 2023-02-05:
It can happen that some scanners don't always scan the exact page size or dpi you specify, so by trying to "correct" it you could end up making it worse.

But I'm going to leave this issue open as it's probably possible to autodetect some cases.

Note to self: this should be easier after implementing dpi capabilities so we have a better idea of the real dpi scanned.