cyanfish/naps2 #72: UWP OCR

lollita, 2023-02-04:
Replace Tesseract dependencies using [UWP OCR](https://docs.microsoft.com/en-us/uwp/api/windows.media.ocr) in windows 10 C#.

cyanfish, 2023-02-04:
Is there a quality or performance benefit for this OCR? What kind of language selection is there (compared to Tesseract's 100+)?

lollita, 2023-02-04:
avoid Tesseract dependecie and 50 GB. See ShareX.

cyanfish, 2023-02-04:
The Tesseract dependency is 3MB in the base download (7MB extracted), plus the language data which varies, e.g. English is a 12MB download (26MB extracted). I consider that minimal and am much more concerned with functionality.

In any case we can't get rid of the Tesseract dependency as we support Windows 7, not to mention Mac and Linux.

closed 2023-02-04 by cyanfish