cyanfish/naps2 #74: Images scanned with 8bpp or 1bpp are always exported as 24bpp in all formats (JPG, PNG, TIFF)

AltoRetrato, 2023-02-05:
**Describe the bug**
No matter the color depth used to scan an image (1bpp, 8bpp, 24bpp), when saved from NAPS2, in any file format, it will always be 24bpp.

**To Reproduce**
Steps to reproduce the behavior:
1. Scan images with different profiles (1bpp, 8bpp, 24bpp)
2. Save the images as JPG, PNG and TIFF.
3. Check the images on another app and check their color depth - they are all 24bpp (even those scanned at 1bpp and 8bpp and saved as PNG or TIFF, which supports those color depths).

**Expected behavior**
All images should be saved in the same color depth they were scanned with.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Version: both in 6.1.2.25834 and 7.0b3

**Additional context**
Add any other context about the problem here.

cyanfish, 2023-02-06:
This should be fixed in [7.0b4](https://www.naps2.com/download) at least for PNG.

AltoRetrato, 2023-02-06:
That's great! I tested BMP, JPEG, PNG and TIFF outputs with NAPS2 7.0b4 on Windows 11, using 1 bpp, 8 bpp and 24 bpp scans. Except for JPEG, all other file formats produced the correct output.

In the case of JPEG, the expected output for 1 bpp and 8 bpp images would be a grayscale JPEG (ImagePixelFormat.Gray8 ?).

cyanfish, 2023-02-07:
Thanks for testing!

For Windows, Grayscale JPEG support is a system limitation that may be addressed by #81. Mac appears to be working with all filetypes. Linux should be working with TIFF only, and fixing that would require directly interfacing with libpng/libjpeg which is beyond the scope of what I want to do though I'll make a todo note in the code.

Anyway, I'm going to close this issue as I don't think there's anything else I plan to do for it for now (aside from #81).

closed 2023-02-07 by cyanfish