cyanfish/naps2 #78: Option to clear workspace after save

lanwin, 2023-02-06:
**Is your feature request related to a problem? Please describe.**
Scanning and saving as PDF tooks some time. So most of the time I click scan or safe and then switch to some other app while its doing its job. 
When I come back I often did not know If I already save the scan as PDF or not cause there is no visual indication.

**Describe the solution you'd like**
It would be helpful if there where an option to clear the workspace after its done. 

**Describe alternatives you've considered**
An alternative would be to add visual that tells me that its already saved. In example there could be an PDF Icon on each side that got saved to PDF.

cyanfish, 2023-02-06:
I want to make it easier to use at some point, but there is an option to do this if (on Windows) you edit the appsettings.xml file in your NAPS2 installation folder and set DeleteAfterSaving to "true".
https://www.naps2.com/doc/org-use#delete-after-saving

lanwin, 2023-02-08:
Uh thank you. I did not noticed this even exists. Works fine.

Should I leave this open as reminder for an Option in the Option Dialog?

lanwin, 2023-09-15:
@cyanfish is there any way to make this setting persist over app updates? Since the appsettings.xml is always overrided, I need to edit it after every update.

cyanfish, 2024-01-15:
In [NAPS2 7.3.0](https://www.naps2.com/download) I've added a "Settings" window with a "Clear images after saving" option for this.

closed 2024-01-15 by cyanfish