cyanfish/naps2 #80: colour pdf size - compression

McFisch1337, 2023-02-06:
**Is your feature request related to a problem? Please describe.**
The sizes of my scanned pdf in s/w are very good (small) but in some usecases i have to scan in colour.
A 2-sided 24bit pdf with qualitiy 40% is round about 1,5 MB. The same pdf scanned with Brother iPrint&Scan is  235kb
I´m using a Brother 4700w at the moment.

**Describe the solution you'd like**
Is there a possibility to add some pdf compressions like mrc? or to change the bitrate to 12bit. This could minimize the pdf´s.
In the office we have a large MFC from Canon, the scanner makes 24bit pdf round about 150kb per page.

**Additional context**
I use NAPS daily and love it. Hope you have a solution for the large files.

cyanfish, 2023-02-10:
Can you attach a sample PDF from Canon + NAPS2 for comparison? There are a number of ways to compress a PDF and I'd be curious to see what's used.

christianclavet, 2023-04-28:
I've implemented this on my fork, from an old pull request from _Luca De Petrillo_ that was sent here years ago. It add JPG codec to the compression and we can change the compression quality.  I had to adapt it since the code was there from a long time and rewrite it. But it work really well.

Was able to get almost a ratio of 5/1 on certain PDF files (75% quality). To propose a new Pull Request I need to get accustomed with this new code since that was done with the 6.1.2 source.

LMSSonos, 2024-03-18:
@christianclavet would you mind to let me know which commit this was in your PR as I would try to make a PR for the current version as it would be great to have this feature, because PDFs with NAPS2 are currently 10x-12x times larger than using acrobat standard

alleluia20, 2024-08-11:
Well, I did read this thread before posting #427, and I am not so sure that my submission was a duplicate. Anyway, I continue replying here, on the open bug,

In #427, I was not pointing out the poor compression of the PDFs generated by NAPS2. What I was telling is that other devices (the Sharp photocopier, in my case) return a PDF with bigger size that NAPS2, but much easier to compress with other tools. Compressing the NAPS2 PDF with the same tools results in much larger size and much worse quality. So **the problem is not (only) compression: the problem is (also) before compression**.

Please read #427 for details.

cyanfish, 2024-08-11:
@alleluia20 There is one thing you can try which is "Document Correction" under the Image menu - that can reduce noise in the image which could potentially make it more compressible by these other tools.

cyanfish, 2026-10-12:
In [8.5.0](https://www.naps2.com/download) there are various improvements to PDF sizes by default. Less OCR overhead, JBIG2 compression, more efficient JPEG compression. There's also a "Compress" function under the Image menu that lets you change image quality after scanning/importing.

closed 2026-10-12 by cyanfish