cyanfish/naps2 #84: Feature Request: Manual two-sided printing 'reorder' option

mrfaptastic, 2023-02-08:
I had to scan about 48 pages the other day which were double sided. Fortunately, I have a scanner that has an automatic document feeder (ADF) but can NOT scan both sides automatically.

So what I did was simply feed in the ADF front side top first to scan pages 1, 3, 5, 7, etc.

Then flip over the stack of paper and scan the other side top first... pages 2, 4, 6, 8 etc.

Perhaps I'm an idiot, but I couldn't find a way to automatically then re-order in NAPS2, which I thought would be an incredibly useful feature to have - simply add an option to the 'reorder' called 'Double sided scan'?  The existing options were of no use to me, or perhaps I have no idea what I'm doing (the naming of them doesn't seems intuiative to be honest).

Basically add a feature to turn the order of scanned pages in NAPS2
```
1, 2, 3, 4, 5, 6, 7, 8, 9
(where 1, 2, 3 are actually pages 1, 3, 5 in the phyiscal document)

to 

1, 6, 2, 7, 3, 8, 4, 9, 5

```

The constraint being, a person must first scan all 'front side' pages, then 'back side' pages, and not then mess with the order in NAPS2.

mrfaptastic, 2023-02-08:
By the way, thanks for making such a great app. Much better than the junk that gets shipped with the usual printers and scanners.

cyanfish, 2023-02-08:
The "Alternative Interleave" option should do this. Though it definitely could be described better.

mrfaptastic, 2023-02-08:
Ah ok. Yeah, it's not obvious, and also, once I had tried using the wrong reorder, it as too late as everything had already been messed up.

grandmastermarco, 2023-02-18:
it would be awesome to be able to preselect which type of reodering the program should do (some kind of default option).
i always forget which type of reorder i need for my particular scanner ;)

TheRealYasri, 2023-05-12:
I just helped my mother do this for her work(nonprofit). If you scan like the OP said then...

select the back 50% of pages scanned. So if its 20 pages, you select 11-20.
reorder -> reverse -> selected 
reorder -> Interleave

It puts the scans in the same order as duplex scan would. Would be nice to have a dialog to do this for the scanner, if it does not support it.

cyanfish, 2023-05-12:
@TheRealYasri "Alternative Interleave" will do this in one go.

TheRealYasri, 2023-05-12:
@cyanfish Yeah your right, thought I tried that reorder combination when I was trying them all to see what they did. Maybe a tooltip to explain it. 

Reorders all scans (Example has 20 scans)
1, 20, 2, 19, 3, 18, 4, 17, etc

cyanfish, 2024-02-05:
Tracking "Manual Duplex" rework in #303

closed 2024-02-05 by cyanfish