cyanfish/naps2 #86: 7.0b4 - crop doesn't retain size between invocations

srdlgh, 2023-02-10:
In the previous Windows incarnation, crop retained the positions between invocations -- which was handy when processing multiple pages of the same size.  Perhaps not doing so now is a 'feature' of the new crop design, but if not it would be nice if this could be fixed.

Different strokes for different folks, and it might be a coding question rather than aesthetics, but the older crop method seemed easier (in particular, required 1 fewer eye-hand coordination steps for corner rectangles, 2 for those not attached to a corner).

cyanfish, 2023-02-10:
Losing retained position wasn't intentional, I'll try and fix that in the next version.

I'm not sure I see how the older method is better - if you drag the handles in the middle of the edges it should be equivalent to dragging the sliders in the previous version, right? The newer version just puts the interaction points on top of the image and also gives you the extra option of dragging the corners.

srdlgh, 2023-02-10:
Ben,

Again, thanks for all the work you've done -- and your responses to my
queries.

It's a subtle part of UI design that precision hand-eye coordination
demands significantly more from the user than an operation that can be left
to lower levels of attention.  In the Windows version, if you have an image
and you pick the crop operation, if you want a region that ends in a corner
you make one precision click in the viewport at the opposite corner of the
rectangle, then zoom to the other end with overshoot (no precision).
Similarly, to select a rectangle that isn't attached, you make a precision
click at one corner and a precision click at the other end of the region.
With the Linux version, you have to make a precision click to pick up the
corner bracket, then another precision click to leave it at the correct
spot, and twice that if the region doesn't end in a corner.

I agree if you use the sliders, the result is similar to the brackets --
but at least so far, it seems to be the sliders require less precision to
grab them than the brackets.

Perhaps the Linux version could get that direct manipulation capability as
well as the brackets.  But either way, I'm in the process of switching over
to NAPS2.

--Steve R

cyanfish, 2023-02-10:
Okay, I see what you're saying. I'm not sure if I'll add another paradigm (maybe after other priorities), but I can definitely make the handles a lot easier to grab.

cyanfish, 2023-03-06:
In 7.0b5 I believe I have fixed the retaining size issue, also made the crop handles bigger / easier to click, and added "click and drag" functionality (when you don't click on a handle).

closed 2023-03-06 by cyanfish