cyanfish/naps2 #89: 7.0b4 Linux - Revert and Reset apparently not working yet

srdlgh, 2023-02-11:
Maybe a bug -- or maybe just something that hasn't made it to the top of the to-do list yet.

cyanfish, 2023-02-11:
Can you be more specific? They seem to be working for me.

srdlgh, 2023-02-13:
Ben --

It looks like I should have sent that report to the Journal of
Irreproducible Results -- it does seem to work.  Possibly I munged it
somewho working on the macros.  I'll let you know if it appears again.

I think I can reproduce this one, though.  Saving as PDF in 7.0b Linux
doesn't seem to add the ".pdf" suffix automatically if not already present.

--Steve R

closed 2023-03-06 by cyanfish

cyanfish, 2023-03-06:
Extensions should now be added when you click Save in the latest [beta version](https://www.naps2.com/download) (7.0b5).