cyanfish/naps2 #99: pdf to odt conversion (retaining text formatting)

kintaro1981, 2023-02-23:
Hello,

is pdf to odt conversion (retaining text formatting) feature going to be included in NAPS2 in the next future?

cyanfish, 2023-02-23:
Hi, I don't currently intend to add support for other filetypes like .docx or .odt. Why do you want this in NAPS2?

kintaro1981, 2023-02-24:
Because I would like to adopt NAPS2  as main app to convert from pdf avoiding to use closed source or online services. I think that NAPS2 in combination with PdfSam (to join/split etc. pages) can became “the pdf solution”.

Right now we (a medium Italian municipality) are already using NAPS2 to convert scanned pdf into searchable and readable from screen readers for impaired eye people and we are trying to find, without success, an open-source project able to convert scanned (and not scanned) pdf to odt retaining text formatting.

Thank you for your time.
Regards

cyanfish, 2023-02-24:
Have you tried using .docx as an intermediary format?
https://github.com/dothinking/pdf2docx
https://stackoverflow.com/questions/21845789/how-to-convert-docx-to-odt-with-libreoffice-on-ubuntu-bash

cyanfish, 2023-02-25:
In any case, while it would be nice in principle to have support for more formats, this is beyond the scope of what I'm planning on building into NAPS2.

closed 2023-02-25 by cyanfish