cyanfish/naps2 #100: 2400dpi support?

feraligatr, 2023-02-25:
The maximum resolution supported in NAPS2 is 1200dpi currently. Could you add 2400dpi?
You might wonder why 2400dpi is needed in the hell. In fact some cheap scanners fake the resolutions so high resolution should be selected to get a relatively clear result. (e.g. I can get 600dpi result when I choose 2400dpi, and about 150dpi when I choose 600dpi)

cyanfish, 2023-02-25:
Thank you for bringing this up, I haven't thought about it lately but it's actually something I should be able to do soon thanks to recent improvements I did to TWAIN.

closed 2023-03-06 by cyanfish

cyanfish, 2023-03-06:
2400 and 4800 dpi should be working in the latest [beta version](https://www.naps2.com/download) (7.0b5).