cyanfish/naps2 #106: Execute Script after PDF saving

torwag, 2023-02-28:
**Is your feature request related to a problem? Please describe.**
I would like to send the generated PDFs directly to certain services like paperless-ngx (API call), SFTP servers, or nextcloud (webdav).

**Describe the solution you'd like**
Similar to #70 but simpler, I would like to propose that a script gets executed (configurable in the GUI) whenever a PDF was generated. 
That could be used to create all kind of "send to..." solutions and more.
The script can be selected in the UI, together with a set of placeholders for certain parameters of naps2, foremost the filename of the PDF. It would be the task of the script maintainer to make use of those parameters. Others could be PDF author, scanner, scan settings, etc. 
The proposed solution is pretty straight forward and yet it could be used for many different situations (send email, send to services, trigger notification, further processing the PDF, send to other devices, print, encrypt, etc.),  enabling NAPS2 to remain slim and not over-featured. 

**Describe alternatives you've considered**
This could be done in an overall script calling naps2 within that script, however it would not work with the batch mode of naps2.
A watchdog could observe a certain folder for incoming PDFs and e.g. send them to a certain API, but that is usually quite fragile and not compatible with all OSes.

SanaRinomi, 2023-12-05:
+1 on this, mainly for the use case of sending to paperless!

gouthamravee, 2024-02-06:
+1 for this, I recently setup paperless ngx and would love to be able to send scans from NAPS2 directly to paperless. 

Their API is described here 

https://docs.paperless-ngx.com/api/#file-uploads
https://docs.paperless-ngx.com/api/#authorization