cyanfish/naps2 #107: Brightness and contrast setting not working

anddam, 2023-03-05:
**Describe the bug**
Changing the `Brightness` and `Contrast` settnigs in a profile results in no change in the scanned image.

**To Reproduce**
Steps to reproduce the behavior:
1. Click on 'Profiles' toolbar button
2. Click on 'Edit' toolbar button to edit the first profile
3. Set brightness and contrast to `-1000` (lowest value)
4. Close profile editing panel, ensure first profile is selected in `Scan` toolbar item
5. Make a scan, the first page appears in current window
6. Click on 'Edit' toolbar button to edit the first profile
7. Set brightness and contrast to `+1000` (highest value)
8. Close profile editing panel, ensure first profile is selected in `Scan` toolbar item
9. Make a scan, the second page appears in current window
10. Note the two images are identical

**Expected behavior**
Two images with very different contrast and brightness

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Version 6.1.2.25834 (latest at the moment of writing)

cyanfish, 2023-03-09:
You can select "Apply brightness/constrast after scan" under Advanced in your profile settings to bypass the scanner's brightness/contrast capabilities (which sometimes don't work).

At some point I want to make this the default behavior so it doesn't need a separate option.

anddam, 2023-03-18:
Confirmed working thanks.

What do the `-1000` and `1000` value means? I mean why the scale to a thousand, positive and negative.

closed 2023-03-18 by anddam