cyanfish/naps2 #108: Auto Rotate

nemchik, 2023-03-08:
**Describe the solution you'd like**
When scanning pages, if each page could be automatically rotated to correct the orientation. An option to enable/disable autorotate based on the scan profile would be ideal.

**Describe alternatives you've considered**
Manually reviewing all scanned pages and rotating them (time consuming and error prone when bulk scanning).

**Additional context**
Occasionally I have staff scanning in bulk and a stack of papers will be placed in the scanner upside down. If the whole stack is upside down it is easy enough to use the select all and rotate option (if the staff notices). Sometimes the contents of the stacks contain papers that are horizontally oriented (rather than vertically), and these can be a little harder to spot (when bulk scanning).

My main concern with the orientation/rotation is the pages will not OCR correctly if they are not correctly rotated. If a rotation detection method is needed, maybe the ORC engine could be leveraged to see if text is readable or not on the page, and if not, try another orientation? That may not be efficient, but I'm just spitballing ideas.

zdenop, 2023-04-21:
As you use Tesseract for OCR, it supports page orientation see: https://github.com/charlesw/tesseract/issues/543
Or you can also use [leptonica](https://github.com/DanBloomberg/leptonica/blob/master/README.md) (the tessereact dependency) for (not sure if you can directly use C library or you need to wrap it):

- https://tpgit.github.io/Leptonica/flipdetect_8c_source.html
- example programs:
  - [prog/livre_orient.c](https://github.com/DanBloomberg/leptonica/blob/5aaf1c187deeef7f47288c6b0833a07021940da7/prog/livre_orient.c)
  - [prog/flipdetect_reg.c](https://github.com/DanBloomberg/leptonica/tree/b21f8e9a3d0c0be402d836045a269676d7408966/prog)

christianclavet, 2023-04-28:
Currently, this is only supported from the scanner itself. If you use the native interface in the scanner profile and your scanner support it, it would rotate the image based on the text written on the page. It's a setting in your scanner driver. (If it support this feature, as not all scanners support it)

Cheb57, 2023-10-11:
Hello @christianclavet, as you mentionned not all scanner native interface or driver are supporting this feature. My concern is that OCR is simply not working if text are not horizontal. If the scanned document is vertically the page will be not search textable using the OCR. Does it possible to use Tesseract dependency libs using the new feature mentionned @zdenop ? Thank you.

christianclavet, 2023-12-16:
@ZeroDot1 and @Cheb57. If you need the OCR to work correctly, you need to only do it when SAVING the document. 

Document scanning is done in at least 2 passes (I work as a scanning tech) and do this with bigger software as Kodak Capture Pro and Iris Document server.  We do thousands of documents boxes, and each batch can get over 1000 pages and more...

Naps is my hobby project and could help in a lot of those tasks. It's not really used where I work mostly because of contracts, but It has the potential to. I try to add features in it so it become more and more able to do the work the "big guys" are doing.

Here what we do, and you can apply this to NAPS too.

1.  You need to prepare your documents (no clips, no bent corner) 
2.  The scan the document per batch in the scanning software. Check for bent corner and doubles.  
3.  Then you do a quality control pass as redo bad pages and **turn the orientation**
4.  Then you export the DOCUMENT as PDF, and the OCR pass is done there.

Doing the OCR while scanning is wrong as it make the scanning pass slow, and hinder productivity. It's MUCH better to do it once the work is done with the quality check.

The feature that @zdenop  is talking is great and it could allow to define OCR Zone that are not horizontally. As an example you have plan of a house and you want the OCR to get the text that is vertically oriented BUT also want to get the text that is oriented horizontally. 

It was never meant as a feature to READ any text in any orientation. You must orient it manually first! Some scanners provide helper to detect the first elements of text and orient the page for you, but it's not 100% accurate! A Quality pass if required for any document you scan if you want to do a quality work with ANY software.

zdenop, 2023-12-16:
@christianclavet: I used NAPS for home manual scanning (usually 1-5 pages, in several cases up to 25). In this setup automatic page orientation and possible deskewing would be very helpful - before quality check. 

Another topic is the scanning of old books or official documents with bent and clipped corners - here I can use NAPS only for scanning (but automatic page orientation would be helpful too). Post-processing I need to do in [scantailor ](https://github.com/trufanov-nok/scantailor-universal)

nemchik, 2023-12-16:
I agree that auto-rotate would slow down scanning, and that any auto-rotate feature in any software is a guess. That being said I do think it has value for someone who has been sit in front of a computer with a warehouse worth of file cabinets or boxes of paper to scan in when a business decides to go paperless.

If an auto-rotate feature were implemented as a button/menu action, that would keep the separation of concerns where scanning can remain fast and OCR can remain part of the saving process.

NAPS is perfect for someone scanning a small number of documents where they know the orientation is all the same or have a reasonable expectation of manually rotating a select number of pages (or put the pages in upside-down and rotate via select all once scanned). This request fills the gap where NAPS does not currently accommodate bulk scanning as mentioned above where pages might be squared up well enough to go through a scanner, but might be oriented rather randomly. Sure it can be done by hand before scanning, but it would be lovely if there was a feature that could attempt a guess that might eliminate the manual pre-scan rotation and let the software be the only place you concern yourself with orientation, even when bulk scanning.

Anyway, regardless, NAPS is still the best scanner software I've ever found, so a huge thanks to the developer(s) even if this feature doesn't make the cut.

christianclavet, 2023-12-17:
Hi! Thanks for replying!

A feature that could be created if someone had the idea for the algorithm, Would be to do the auto-rotate as a post-process. This way if you scanner doesn't support it, the software would help you with this. 

Lots of scanners have auto-rotate. I wonder what scanner cannot do this with Naps?  @nemchik What scanner do you use? With that lot of paper, you could get one that has the feature, it would help you a lot, even if it's not perfect...

@zdenop  

> in this setup automatic page orientation and possible deskewing would be very helpful - before quality check.

Deskewing can be done on page selections, on 25 page, would be damn easy to do with that number of pages... You can even enable it in the advanced options of the scanner in naps. You could do in Naps -> Select ALL, then DESKEW. 

Manuscript text and pictures|drawings will NOT be rotated in any way, even if the scanner support auto-rotation. It's not supported by any code. For theses, regardless of having the feature or not, you must rotate it manually.

If you really want a "fire and forget" solution to this with quality, you need to wait to AI development to get into scanning. So it read and UNDERSTAND the page so it rotate it accordingly each time. At the moment, we use tricks, but it's far from perfect.

nemchik, 2023-12-17:
> Lots of scanners have auto-rotate. I wonder what scanner cannot do this with Naps? @nemchik What scanner do you use? With that lot of paper, you could get one that has the feature, it would help you a lot, even if it's not perfect...

Mostly Fujitsu fi series scanners. A various mix of model numbers. I haven't specifically looked for auto-rotate/orientation features within the scanner drivers. First place I looked was in NAPS.

christianclavet, 2023-12-17:
Hi! Thanks @nemchik, try in native mode first and check in the driver of the manufacturer. There should be an option to do this there. We have a Fujitsu fi model that we use for small work. I'll check it out...