cyanfish/naps2 #109: Dark mode support

cyanfish, 2023-03-09:


ace280, 2023-03-11:
Yes please =D

joshuadavidmcgee, 2023-04-03:
+1 for Windows Dark Mode Support

cyanfish, 2023-07-09:
Mac and Linux should now support dark mode in [7.0b9](https://www.naps2.com/download). Windows won't happen for a while as it depends on #81.

Unixware, 2023-07-23:
I think you may missed few parts of the code to change, eg things like the main background , also background of the scanner profiles and  of  the language chooser page (OCR)  are still 'white' , even when the dconf set the org.gnome.desktop.interface.color-scheme  as 'prefer-dark'  
tnx

cyanfish, 2023-07-28:
@Unixware What distro and desktop environment are you using?

Unixware, 2023-07-28:
sorry I should mention it, Gnome (on debian 12)

cyanfish, 2023-09-05:
[7.1.0](https://www.naps2.com/download) has some improvements for dark theme detection that should resolve this.

LinuxOnTheDesktop, 2023-11-28:
Hello!

I've 7.1.2 and . . it's all white on a Windows that is dark-themed.

cloudmaker97, 2024-05-04:
> Hello!
> 
> I've 7.1.2 and . . it's all white on a Windows that is dark-themed.

Yeah, the implementation of this feature it's currently blocked by #81 

@cyanfish Could I assist you with the issue #81 in some way? Or do you plan it to do by yourself? 😊

cyanfish, 2024-10-05:
Windows dark mode support has been added in [8.0b1](https://www.naps2.com/download).

closed 2024-10-05 by cyanfish