cyanfish/naps2 #110: `DeleteAfterSaving == true` does not clear images after saving a PDF (version `6.1.2.25879`)

thehale, 2023-03-13:
**Describe the bug**
`appsettings.xml` -> `DeleteAfterSaving` == `true` does not clear the images after saving a PDF.

**To Reproduce**
Steps to reproduce the behavior:
1. Set `DeleteAfterSaving` to `true` in `appsettings.xml`
2. Open NAPS2
3. Scan a document (1 or more pages)
4. Save document as a PDF
5. Observe that the scanned images are still present.

**Expected behavior**
After saving, I expect the images to be gone so that I can start scanning the next document without needing to press "Clear" and its confirmation dialog.

**Screenshots**
See the scanned images still in the application even though the notification in the bottom right states that they were successfully saved.
[image]

**Desktop (please complete the following information):**
 - OS: Windows 11 Home 22H2
 - Version 6.1.2.25879 (PortableApps)

**Additional context**
Issue #78 references this same appsetting, but focuses on providing a UI option for it rather than reporting a bug with its functionality.

I really appreciate NAPS2 because it is more reliable than my printer manufacturer's scanning software. This bug is my only major pain point since I scan lots of small documents, meaning it quickly becomes tedious to clear the screen after every save. This fix would greatly improve the quality of my experience with NAPS2.

cyanfish, 2023-03-13:
Please try with the latest [beta version](https://www.naps2.com/download).

thehale, 2023-03-15:
That does the trick! Version `7.0b6` clears the images after saving just as expected.

I missed the beta because I was looking at the "Releases" preview on the project's landing page on GitHub, which showed `6.1.2` as the latest release.

closed 2023-03-15 by thehale