cyanfish/naps2 #112: System.Exception: System.EntryPointNotFoundException: Unable to find an entry point named 'dlopen' in shared library 'libc.so.6'.

alicavus, 2023-03-14:
When I try to add SANE scanner I got:
[image]
```
System.Exception: System.EntryPointNotFoundException: Unable to find an entry point named 'dlopen' in shared library 'libc.so.6'.
   at NAPS2.Platform.Linux.LinuxInterop.dlopen(String filename, Int32 flags)
   at NAPS2.Platform.LinuxSystemCompat.LoadLibrary(String path) in /home/ben/naps2/NAPS2.Sdk/Platform/LinuxSystemCompat.cs:line 48
   at NAPS2.Unmanaged.NativeLibrary.DoLoadLibrary(String path) in /home/ben/naps2/NAPS2.Sdk/Unmanaged/NativeLibrary.cs:line 60
   at System.Lazy`1.ViaFactory(LazyThreadSafetyMode )
   at System.Lazy`1.ExecutionAndPublication(LazyHelper , Boolean )
   at System.Lazy`1.CreateValue()
   at System.Lazy`1.get_Value()
   at NAPS2.Unmanaged.NativeLibrary.get_LibraryHandle() in /home/ben/naps2/NAPS2.Sdk/Unmanaged/NativeLibrary.cs:line 71
   at NAPS2.Unmanaged.NativeLibrary.LoadFunc[T]() in /home/ben/naps2/NAPS2.Sdk/Unmanaged/NativeLibrary.cs:line 81
   at NAPS2.Unmanaged.NativeLibrary.<Load>b__12_0[T]() in /home/ben/naps2/NAPS2.Sdk/Unmanaged/NativeLibrary.cs:line 75
   at NAPS2.Util.CollectionExtensions.Get[TKey,TValue](Dictionary`2 dict, TKey key, Func`1 defaultValue) in /home/ben/naps2/NAPS2.Internals/Util/CollectionExtensions.cs:line 215
   at NAPS2.Unmanaged.NativeLibrary.Load[T]() in /home/ben/naps2/NAPS2.Sdk/Unmanaged/NativeLibrary.cs:line 75
   at NAPS2.Scan.Internal.Sane.Native.SaneNativeLibrary.get_sane_init() in /home/ben/naps2/NAPS2.Sdk/Scan/Internal/Sane/Native/SaneNativeLibrary.cs:line 40
   at NAPS2.Scan.Internal.Sane.Native.SaneClient..ctor(ISaneInstallation saneInstallation) in /home/ben/naps2/NAPS2.Sdk/Scan/Internal/Sane/Native/SaneClient.cs:line 22
   at NAPS2.Scan.Internal.Sane.SaneScanDriver.<>c__DisplayClass8_0.<GetDevices>b__0() in /home/ben/naps2/NAPS2.Sdk/Scan/Internal/Sane/SaneScanDriver.cs:line 56
   at System.Threading.Tasks.Task.InnerInvoke()
   at System.Threading.Tasks.Task.<>c.<.cctor>b__272_0(Object )
   at System.Threading.ExecutionContext.RunFromThreadPoolDispatchLoop(Thread , ExecutionContext , ContextCallback , Object )
--- End of stack trace from previous location ---
   at System.Threading.ExecutionContext.RunFromThreadPoolDispatchLoop(Thread , ExecutionContext , ContextCallback , Object )
   at System.Threading.Tasks.Task.ExecuteWithThreadLocal(Task& , Thread )
--- End of stack trace from previous location ---
   at NAPS2.Scan.Internal.RemoteScanController.GetDevices(ScanOptions options, CancellationToken cancelToken, Action`1 callback) in /home/ben/naps2/NAPS2.Sdk/Scan/Internal/RemoteScanController.cs:line 23
   at NAPS2.Remoting.Worker.WorkerServiceImpl.GetDevices(GetDevicesRequest request, IServerStreamWriter`1 responseStream, ServerCallContext context) in /home/ben/naps2/NAPS2.Sdk/Remoting/Worker/WorkerServiceImpl.cs:line 117
   at NAPS2.Remoting.RemotingHelper.HandleErrors(Error error) in /home/ben/naps2/NAPS2.Sdk/Remoting/RemotingHelper.cs:line 24
   at NAPS2.Remoting.Worker.WorkerServiceAdapter.GetDevices(ScanOptions options, CancellationToken cancelToken, Action`1 callback) in /home/ben/naps2/NAPS2.Sdk/Remoting/Worker/WorkerServiceAdapter.cs:line 56
   at NAPS2.Scan.Internal.WorkerScanBridge.GetDevices(ScanOptions options, CancellationToken cancelToken, Action`1 callback) in /home/ben/naps2/NAPS2.Sdk/Scan/Internal/WorkerScanBridge.cs:line 27
   at NAPS2.Scan.ScanController.<>c__DisplayClass12_0.<<GetDevices>b__0>d.MoveNext() in /home/ben/naps2/NAPS2.Sdk/Scan/ScanController.cs:line 61
--- End of stack trace from previous location ---
   at NAPS2.Util.AsyncProducers.<>c__DisplayClass1_0`1.<<RunProducer>b__0>d.MoveNext() in /home/ben/naps2/NAPS2.Internals/Threading/AsyncProducers.cs:line 22
   at NAPS2.Util.AsyncSink`1.AsAsyncEnumerable()+MoveNext()
   at NAPS2.Util.AsyncSink`1.AsAsyncEnumerable()+System.Threading.Tasks.Sources.IValueTaskSource<System.Boolean>.GetResult()
   at NAPS2.EtoForms.Ui.SelectDeviceForm.ShowAsyncDeviceList() in /home/ben/naps2/NAPS2.Lib/EtoForms/Ui/SelectDeviceForm.cs:line 66
   at NAPS2.EtoForms.Ui.SelectDeviceForm.ShowAsyncDeviceList() in /home/ben/naps2/NAPS2.Lib/EtoForms/Ui/SelectDeviceForm.cs:line 66
```
I tryed to build the package myself I've got same error.

ceandre, 2023-03-14:
Same here...
Rocky Linux 9
[image]

ceandre, 2023-03-15:
> Same here... Rocky Linux 8.7 [image]

Same on CentOS 7.9....
7.0.5b don't give this error but also don't find any scanner (scanimage -L it's ok)

cyanfish, 2023-03-15:
Can you try this version and see if it works? [naps2-7.0b6-test.rpm.gz](https://github.com/cyanfish/naps2/files/10975830/naps2-7.0b6-test.rpm.gz)

cyanfish, 2023-03-15:
Or if you prefer to build from source you can try this [patch](https://github.com/cyanfish/naps2/files/10975847/dlfix.patch).

ceandre, 2023-03-15:
[image]
Hey! 
Working now on Rocky 8.6! Thanks!  I'm going to check it on CentOS 7.x and let you know. :D

ceandre, 2023-03-15:
Now it's working on CentOS 7.9! Thanks again!

closed 2023-03-20 by cyanfish

cyanfish, 2023-03-20:
This should be fixed in the [latest beta (7.0b7)](https://www.naps2.com/download).

commit 2a1e922b (2023-03-20): Use libdl for library loading on Linux

Fixes #112