cyanfish/naps2 #115: NAPS2 using variable sequence number to identify scanners

ceandre, 2023-03-15:
NAPS2 using variable sequence number to identify scanners.
Airscan use protocol first letter plus sequence number of discovery on scanner identification, it's ok on Airscan, but on NAPS we have a problem since we can save scanning profile. 

Because of this airscan behavior If first scanner of discovery got offline, then all another scanners get another sequence number, then all configured profiles won't work anymore on NAPS2

Example:

On NAPS source selection I got:
---------------------------------
Lexmark Management (airscan)
Lexmark Dev (airscan)
Lexmark Entrance (airscan)
---------------------------------

Then I create 3 profiles, one for each.

For a now all three profiles are working nice, but if first printer get offline, next time, another two profiles wont work anymore.
On profiles.xml, we got (eSCL protocol):
<ID>airscan:e0:Lexmark Management</ID>
<ID>airscan:e1:Lexmark Dev</ID>
<ID>airscan:e2:Lexmark Entrance</ID>

but if first scanner get offline... then sequence protocol+number+number chances to:

airscan:e0:Lexmark Dev
airscan:e1:Lexmark Entrance

Then ALL saved NAPS2 profiles stop working.

There any way to avoid this problem? We can just use printer name as ID without airscan sequence?

Thanks.

cyanfish, 2023-03-16:
Does the "escl" (not "airscan") backend show as an option for you? It seems like it has a more consistent id.

ceandre, 2023-03-16:
> Does the "escl" (not "airscan") backend show as an option for you? It seems like it has a more consistent id.

In fact, for Lexmark scanner Airscan find both, WSD and eSCL (on scanimage -L), but you mean "native" eSCL backend from newer sane-backends releases? if so, on CentOS 7.x/8.x sane-backends doesn't have native eSCL implententation (don't even have sane-airscan, but I as able to find a repository with compatible rpm package)

cyanfish, 2023-03-17:
Yeah, that's what I meant. I'll see if I can find a solution for the Airscan issue, but in the meantime if you want you can also try the [flatpak](https://www.naps2.com/download) which should include the escl backend.

cyanfish, 2023-12-13:
In [7.2.0](https://www.naps2.com/download) there's now a built-in "ESCL Driver" option (select in your profile settings). This is now the recommended way to connect to ESCL scanners, and shouldn't have the same problem.

The ESCL Driver should work as well or better than Airscan, but if you have any issues with it please do open a new issue.

closed 2023-12-13 by cyanfish

cyanfish, 2024-02-05:
The airscan issue should also be fixed in 7.3.1.

cyanfish, 2025-07-08:
@alexpevzner Fair point regarding firmware workarounds, but users can certainly use still use AirScan if that's applicable. I'm continuing to bundle it with the flatpak as well.

The benefits I'm referring to are mostly in regard to the SANE protocol and not AirScan itself, namely:
- Showing devices immediately (rather than having to wait for all SANE backends to complete)
- Progress reporting during a scan
- Showing the device-provided icon
- "Stickiness" to the known-working IP (in case of temporary mDNS issues) as well as being able to manually specify an IP

That said, I would be very happy to integrate an AirScan UUID option if you choose to add it.