cyanfish/naps2 #118: DPI setting does not seem to work

anddam, 2023-03-18:
**Describe the bug**
Profile DPI setting does not seem to be taken into account.

**To Reproduce**
Steps to reproduce the behavior:
1. Create two proifiles, A4, 24 bit depth, one 300 DPI the other 1200 DPI
2. Make two scans of the same imaeg
3. Check the size of the obtained files are wildly different

**Expected behavior**
I expected the saved PDF be very different in quality and size on disk. The visual quality looks pretty much the same zooming at 800% on a detail, upon inspecting the PDF files the contained image obejct are very similar, 707739 and 710281 bytes for 300 and 1200 DPI resp.

**Screenshots**
Compared view at 800% zoom
[image]

**Desktop:**
 - OS: Windows 11
 - Version 6.1.2.25834

cyanfish, 2023-03-19:
What model scanner do you have and what's the max dpi it advertises?

Try switching between WIA and TWAIN drivers and see if the other works better.

Also if you're using WIA there's a bug where if you select too high a DPI it will use the default DPI (instead of the highest available). So you can try 600 or use the latest [beta](https://www.naps2.com/download) where that bug is fixed.

closed 2023-03-24 by cyanfish

anddam, 2023-08-27:
A bit late on this, but just as update: I have an *HP LaserJet Pro MFP M28w* that is being used over WiFi. The [hardware specs](https://support.hp.com/us-en/document/c05968109#AbT14) mention 600 dpi × 600 dpi and "TWAIN 2.1".

I can only use the scanner using WIA driver on NAP2, trying to select TWAIN pops up a source selection dialog with options:

* HP LJ M28M31 TWAIN
* HP TWAIN USB

The former is likely the one to be used, given it has the device name and the latter has "USB" but in both cases I get a splash screen "Waiting for TWAIN to complete…" and then an HP error dialog "The HP LaserJet MFP M28-M31 was not found."

I think that is trying to use an USB connection, that is not there at all.

Comparing the same image acquired at 300 and 100 dpi shows vast difference in the image quality so the issue was not there in first place.