cyanfish/naps2 #120: Can I call the scanner using javascript，thanks.

WeiLu67, 2023-03-23:
Is your feature request related to a problem? Please describe.
Due to many scenarios, it is necessary to call the scanner on the H5 page and obtain the corresponding blob file stream for display on the H5 page.

Describe the solution you'd like
It can be run as a background service (mainly supporting Windows) and supports HTTP requests,
such as **[scanx](https://github.com/balbarak/scanx)** (This project only supports WIA and is no longer maintained)

Additional context
I tried Naps and felt that it was the first and most powerful source in open source. I hope it can support more docking methods.

cyanfish, 2023-03-23:
[NAPS2.Sdk](https://github.com/cyanfish/naps2/tree/master/NAPS2.Sdk) is a work in progress that will provide a .NET library for scanning. This could be used to build a service like that.

WeiLu67, 2023-03-24:
Thanks for you reply. So, may I take the liberty to ask if this is in the future plan

cyanfish, 2023-03-24:
It's not in my plans - maybe someday but there are a lot of higher priority things for me.

If anyone else wanted to build such a thing on top of NAPS2.Sdk I'd be happy to help.

WeiLu67, 2023-03-24:
Thanks，i got it.

yelhouti, 2023-08-25:
@cyanfish I want to start working on a javascript package that would call the SDK using http(s) use .Net Core.

I have written at most 10 lines of codes of .Net/C# to contribute PRs for small bugs in some projects. But never a full project. Since the nugget for the SDK is not yet available, I am wasting time trying to link my local project to yours.
If you could help me with all this I would be very greatful.

Also, now that I read that you are interested in this. Maybe I can make a PR with new subProject Naps.WebServer that calls the SDK. Maybe the project setup would be easier.

yelhouti, 2023-08-26:
@cyanfish I managed to create a controller in project with code like this:
```
using Microsoft.AspNetCore.Mvc;
using Microsoft.Extensions.Logging;
using NAPS2.Images.Gdi;
using NAPS2.Scan;
using NAPS2.Images;

namespace NAPS2.WebApi.Controllers
{

    [ApiController]
    [Route("api/[controller]")]
    public class ScannerController : ControllerBase
    {

        private readonly ILogger<ScannerController> _logger;

        private readonly ScanController _scanController;

        public ScannerController(ILogger<ScannerController> logger)
        {
            _logger = logger;
            // TODO depending on OS use GDI/GTK... (this only works for windows)
            using ScanningContext scanningContext = new ScanningContext(new GdiImageContext());
            _logger.LogInformation("scanningContext: " + scanningContext);
            _scanController = new ScanController(scanningContext);
        }

        [HttpGet("list")]
        public async Task<ActionResult<List<ScanDevice>>> ListDevices()
        {
            // TODO based on env Var change driver
            // TODO set 32 or 64 based on OS
            var devices = await _scanController.GetDeviceList(new ScanOptions { Driver = Driver.Twain, TwainOptions = { Dsm = TwainDsm.NewX64 } });
            return Ok(devices);
        }

        [HttpGet("scan/{deviceId}")]
        public async Task<ActionResult<List<ProcessedImage>>> Scan([FromRoute] string deviceId)
        {
            var devices = await _scanController.GetDeviceList(new ScanOptions { Driver = Driver.Twain, TwainOptions = { Dsm = TwainDsm.NewX64 } });
            ScanDevice device = devices.Find(d => d.ID == deviceId);
            if (device == null)
            {
                return NotFound();
            }
            var scannedImages = new List<ProcessedImage>();
            await foreach (var processedImage in _scanController.Scan(new ScanOptions { Device = device, Driver = Driver.Twain, TwainOptions = { Dsm = TwainDsm.NewX64 } }))
            {
                scannedImages.Add(processedImage);
            }
            return Ok(scannedImages);

        }
    }
}

```

The issue I am facing with SDK is that

```
 await _scanController.GetDeviceList(new ScanOptions { Driver = Driver.Twain, TwainOptions = { Dsm = TwainDsm.NewX64 } });
```
never finishes, any idea why ?.

Of course this code should be improved and some options should be passed in POST body on scan...

cyanfish, 2023-08-26:
I've published a first [alpha version](https://www.nuget.org/packages/NAPS2.Sdk) of the SDK so you should be able to use that. There might be an issue though where you might need to set `ScanOptions.TwainOptions.Dsm = TwainDsm.Old` - see if that makes a difference.

yelhouti, 2023-08-26:
@cyanfish I am getting:
> Tried to run TWAIN from a 64-bit process. If this is intentional, set ScanOptions.TwainOptions.Dsm to TwainDsm.NewX64. Otherwise you can set up a worker process with ScanningContext.WorkerFactory.

Since you suggested not to use `TwainDsm.NewX64`, I tried creating the `ScanningContext.WorkerFactory` but using it is not straight forward. Could help me with a full exemple on how to use `TwainDsm.Old`. Thanks in advance

cyanfish, 2023-08-26:
Sorry yes that message needs updating. I would recommend using the nugets, not sure how easy it would be to get the exe working otherwise.

Instructions are here https://github.com/cyanfish/naps2/blob/master/NAPS2.Sdk.Samples/TwainSample.cs

yelhouti, 2023-08-26:
I tried the sample code with nuggets it fails because of the bug mentioned before:
>  Tried to run TWAIN from a 64-bit process. If this is intentional, set ScanOptions.TwainOptions.Dsm to TwainDsm.NewX64. Otherwise you can set up a worker process with ScanningContext.WorkerFactory.

When I try to use my code using the factory and the nuggets, it doesn't work because `NAPS2.Remoting.Worker` is not published.
```
            scanningContext.WorkerFactory = WorkerFactory.CreateDefault();
            scanningContext.WorkerFactory.Init(scanningContext);
```

Could you provide a project that works with nuggets, or a sample with Main that i can easily run ?
[https://github.com/cyanfish/naps2/blob/master/NAPS2.Sdk.Samples/](https://github.com/cyanfish/naps2/blob/master/NAPS2.Sdk.Samples/TwainSample.cs) doesn't have a main so i don't know how to start them (again not to much experience with C#) ?

cyanfish, 2023-08-26:
If you look at the sample I linked it explains, see here:
https://github.com/cyanfish/naps2/blob/master/NAPS2.Sdk.Samples/TwainSample.cs

Specifically you just need to call `scanningContext.SetUpWin32Worker()`.

yelhouti, 2023-08-26:
@cyanfish thanks a lot for adding that sample it really helped, I was able to scan documents. The only thing left now is to export them as jpeg.

How do I make sure to always return scan in JPEG ?

cyanfish, 2023-08-26:
That should work but I would recommend this (which avoids using the GDI-specific Bitmap type):

```
private string processedImageToBase64(ProcessedImage processedImage)
{
    var stream = processedImage.Render().SaveToMemoryStream(ImageFileFormat.Jpeg);
    return Convert.ToBase64String(stream.ToArray());
}
```

I'll also mention that you may or may not want to use base64; certainly it's a reasonable option (and works nicely with JSON), but for large images it might be more efficient to [return the binary data](https://stackoverflow.com/questions/48653662/how-do-i-return-a-byte-array-from-an-action) and have the Javascript read it [as a blob](https://developer.mozilla.org/en-US/docs/Web/API/Response/blob).

yelhouti, 2023-08-26:
Thanks for all the tips, but how would I return multiple blobs ?

cyanfish, 2023-08-26:
Yeah that would be tougher, you might need to do something like have the server be stateful (e.g. store the images or scanning state in a static session variable and have a separate action call to get each image).

Anyway, don't let me stop you from doing the base64 way if that's easier to get something working, maybe could do the other way later down the road.

yelhouti, 2023-08-26:
@cyanfish I have a working server now, do you want me to make a PR for Naps2.WebApi so that other can contribute, make it cross paltforms, allow more params...

Or should I create a new project ?

cyanfish, 2023-08-26:
I would recommend creating your own project.

yelhouti, 2023-08-27:
For anyone interested here is a working server for windows/TWAI, PR as more than welcome for other drivers/OSs.

I will add a README and releases later. For now it can be build locally into an MSI and installed wherever you want.

https://github.com/tailosoft/scan-server

cyanfish, 2023-08-27:
Cool, that's great! And good to see the SDK is already getting some use.

I would suggest adding a license file to the project. If you like you can use the [LGPL](https://github.com/cyanfish/naps2/blob/master/NAPS2.Sdk/LICENSE) which NAPS2.Sdk uses.

I also just published [NAPS2.Images.ImageSharp](https://www.nuget.org/packages/NAPS2.Images.ImageSharp) so you can use `ImageSharpImageContext` instead of `GdiImageContext` if you like - that will work across all platforms without needing any extra work.

yelhouti, 2023-08-28:
@cyanfish ImageSharp doesn't seem to work in my project, you can try it. It fails to save for same reason because of an unimplemented method if I read correctly.

Also, When a create a windows service and run it, for some reason, the endpoints specific to the scanner hangs indefinitely, I am suspecting it is because of SetUpWin32Worker Could you please have a look ? Thanks in advance

cyanfish, 2023-08-29:
Yeah I see a couple ImageSharp issues, I'll try and fix those at some point. For the service I don't know what the issue is but it's not the worker; scanning with WIA (and no worker) has the same problem.

yelhouti, 2023-08-29:
@cyanfish I found that the issue happens at the gRPC it looks like the server hangs waiting for a response. This looks like some locking issue that happens only when in Windows Service. Your help would be much appreciated

cyanfish, 2023-08-29:
That means the issue is somewhere in the actual scanning process (which happens in the worker), it doesn't really narrow it down past that. It would be easier for you to diagnose the problem without the worker.

Does your scanner work with WIA (instead of TWAIN)? If so that would avoid the worker.

Alternatively you can try building your server exe as 32-bit (i.e. win-x86 instead of win-x64). Then TWAIN would work without the worker.

yelhouti, 2023-08-29:
@cyanfish thanks for pointing me to WIA, I should have written here this morning. You rock

cyanfish, 2023-08-30:
No problem. Also ImageSharpImageContext should work now with 0.1.0-alpha3.

cyanfish, 2024-08-17:
If anyone cares to try there is an [ESCL JS/TS library](https://www.npmjs.com/package/escl-sdk-ts), which should connect natively to NAPS2's [scanner-sharing server](https://github.com/cyanfish/naps2/blob/master/NAPS2.Sdk.Samples/NetworkSharingSample.cs).

cyanfish, 2024-08-17:
This does seem to be working (with NAPS2.Sdk 1.1.2 and a lightly modified version of escl-sdk-ts). I've created the [naps2-webscan](https://github.com/cyanfish/naps2-webscan) project as the canonical sample/documentation for this.

I now consider this as the recommended approach for scanning from a web browser with NAPS2, and I'll close this issue. Feel free to open issues or PRs on the naps2-webscan project.

closed 2024-08-17 by cyanfish