cyanfish/naps2 #124: More detailed OCR in PDF24 than in NAPS2 with the same Tesseract version

NextTherapist, 2023-03-29:
Hi,

I found differences between OCR of PDF24 and NAPS2, both newest versions, both containing Tesseract 5.2.0, when doing OCR on the same black/white PDF file with a gridded background like a slight grey (background from a scanned form).

PDF24 detects small or pixelated characters better, and sometimes NAPS2 detects nothing at all instead. Both need the same amount of time.

Something must be different in the Tesseract integration of both softwares, since both say they use Tesseract 5.2.0. NAPS2 (7.0b7) is on "best" quality, PDF24 has no setting for quality.

Perhaps there is something that could be improved in the NAPS implementation of Tesseract?

cyanfish, 2023-03-29:
I'll need a sample file with the issue to do a comparison.

cyanfish, 2023-04-08:
I did a comparison and couldn't identify a significant quality difference. Feel free to attach a sample file that is significantly better in PDF24 and I'll have a look.

closed 2023-04-08 by cyanfish