cyanfish/naps2 #126: Remember previous session

Germs2004, 2023-04-08:
**Is your feature request related to a problem? Please describe.**
If NAPS2 closes before I've saved all the images, I lose them all and have to re-scan them all.  It could close due to user accidentally clicking the X, NAPS2 crashing, or Windows automatically rebooting for an update.

**Describe the solution you'd like**
NAPS2 should have an option to "remember previous session".  When enabled, if you close NAPS2 while images are still in the workspace, those images will be saved to a NAPS2 folder.  The next time you open NAPS2, it should automatically resume the last session, showing the same images rather than a blank workspace.

**Describe alternatives you've considered**
If any unsaved images are in the NAPS2 workspace when you try to close NAPS2, it could refuse to exit until the user clears the workspace or saves those images.  If NAPS2 knows that all the images have been saved, or were created by importing an existing file, it could allow itself to exit normally.

cyanfish, 2023-04-08:
Most of this already exists. If you try to exit without saving changes you'll be prompted as such. If NAPS2 crashes or closes unexpectedly it will prompt to recover the next time you start it.

Germs2004, 2023-04-08:
My apologies, I didn't know.  I just experienced data loss somehow and had to rescan all my papers, so I came to write the feature request.  I'm not sure if I mindlessly closed NAPS or if it crashed somehow.  That recovery feature sounds great for crashes, but could we make it also work for normal exits as well?  That would allow us to resume the previous session in every situation.  I love how Notepad++ does that, which makes it safe to close any time and resume later.

cyanfish, 2023-04-08:
As far as the data loss, what version of NAPS2 are you using? Did you restart your computer? By any chance did you save some of the data (Save Selected) before it happened?

Germs2004, 2023-04-08:
I scanned 4 documents into the NAPS2 workspace, saved one of them as a PDF, and as I was flipping between windows and closing old windows, I'm guessing I closed the NAPS2 window by accident.  When I noticed NAPS2 was closed, I reopened it, and the 3 unsaved documents were gone.  It was probably just my mistake.  I simply rescanned the 3 other documents and saved them, so all is well.  Then I had the idea of an automatically saving session to prevent that issue in the future.

cyanfish, 2023-04-08:
Okay I understand what happened. NAPS2 doesn't track the individual pages that have been saved, once you saved the first doc it assumed you had saved everything you needed and allowed you to close without prompting. That's definitely something I can improve.

christianclavet, 2023-04-28:
@cyanfish  Your recovery system is really good! If the code in the new version has not changed too much, you could disable the flag when you export your data and add a 3rd button when you close the application to keep the content, and do an action to exit and do not delete anything. 

The next time you will start it, the recovery system will ask you to recover (or not) the last session.
That was quite simple to do with the 6.12 source. And I later implemented to reload any session using a folder request...

cyanfish, 2024-01-15:
In [NAPS2 7.3.0](https://www.naps2.com/download) I've added a "Settings" window with a "Keep images across sessions" option that does this.

closed 2024-01-15 by cyanfish