cyanfish/naps2 #129: Scanning with Duplex and A4 hangs

xM8WVqaG, 2023-04-08:
**Describe the bug**
A clear and concise description of what the bug is.

Scanning with "Paper source" set to "Duplex" and with "Page size" set to "A4" just hangs NAPS2 

**To Reproduce**
Steps to reproduce the behavior:

* Scanning with "Paper source" set to "Glass" or "Feeder" with "Page size" set to "A4" works fine
* Scanning with "Paper source" set to "Duplex" with "Page size" set to "US Legal" or "US Letter" works fine
* Scanning with "Paper source" set to "Duplex" with "Page size" set to "A4" just hangs
  * NAPS2 is still responsive and you can cancel the scan
  * However, after this NAPS2 can't communicate with the scanner again until it's been closed and opened again
    [image]
  * After NAPS2 has closed, the scanner will pass the document through even though there is no software left handling the scan

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Ubuntu 22.04

```
$ flatpak info com.naps2.Naps2
NAPS2 - Not Another PDF Scanner - Scan documents to PDF and more, as simply as
possible.

          ID: com.naps2.Naps2
         Ref: app/com.naps2.Naps2/x86_64/beta
        Arch: x86_64
      Branch: beta
     Version: 7.0b7
     License: GPL-2.0-or-later
      Origin: naps2-origin
  Collection: 
Installation: user
   Installed: 72.9 MB
     Runtime: org.freedesktop.Platform/x86_64/22.08
         Sdk: org.freedesktop.Sdk/x86_64/22.08

      Commit: ef8d9f575f473dc663090f4e0c6a1ca36c2eea0b10bb26d9f3d2b06882fbea90
     Subject: Export com.naps2.Naps2
        Date: 2023-03-20 01:19:51 +0000
```

Same behaviour with the `deb` package

```
$ apt list | grep ^naps
naps2/now 7.0.7 amd64 [installed,local]
```

**Additional context**
Add any other context about the problem here.

xM8WVqaG, 2023-04-09:
This doesn't appear to be an issue specific to the A4 paper size.

I can reproduce the hang using a custom paper size of 2 inches^2.

[image]

After some more debugging, I found even when using "US Legal" (216 x 356) which should comfortably fit A4 (210 x 297) scans were being truncated.

Using [`scanimage`](http://www.sane-project.org/man/scanimage.1.html) to confirm it isn't a hardware issue, I can reproduce the truncation with the default scan options.

```text
$ scanimage -A

All options specific to device `fujitsu:fi-6110dj:1858':
...
  Geometry:
...
    --page-height 0..876.695mm (in steps of 0.0211639) [279.364]
        Specifies the height of the media.
```

By adjusting the `page-height` I was able to capture the full document with:

```shell
$ scanimage --page-height=297.015 -o foo.jpg
```

cyanfish, 2023-04-09:
Interesting, I suspect that the fujitsu backend is relying on page-height/page-width instead of the more standard tl-x/tl-y/br-x/br-y options (that define the X/Y values of the top-left corner and bottom-right corner) that NAPS2 sets. Can you try setting those options with scanimage instead (e.g. br-y=297.015)?

xM8WVqaG, 2023-04-09:
Attempting to set only the `br-y` without first increasing the `page-height` makes it round the `br-y` value down to the default `page-height` and truncates the scan accordingly:

```shell
$ scanimage -y 297.015 -o y-only.jpg
scanimage: rounded value of br-y from 297.015 to 279.364
```

Setting `page-height` to ~US Legal and then setting `br-y` to ~A4 scans the correct area:

```shell
$ scanimage -y 297.015 --page-height=355.998 -o y-and.jpg
scanimage: rounded value of page-height from 355.998 to 355.999
scanimage: rounded value of br-y from 297.015 to 297.015
```

However, that doesn't completely explain the error, as my previous _Very Small_ (2 inches^2) test worked in `scanimage` only setting `br-x` and `br-y` values:

```shell
$ scanimage -y 50.801 -x 50.801 -o very-small.jpg
scanimage: rounded value of br-x from 50.801 to 50.7935
scanimage: rounded value of br-y from 50.801 to 50.7935
```

I have successfully used this scanner with an older version of NAPS2 on Windows 7 >5 years ago, although that would have likely been TWAIN there. So this is probably only an issue for SANE.

For reference, here is the complete output of `scanimage -A`:

```text
All options specific to device `fujitsu:fi-6110dj:1858':
  Standard:
    --source ADF Front|ADF Back|ADF Duplex [ADF Front]
        Selects the scan source (such as a document-feeder).
    --mode Lineart|Halftone|Gray|Color [Lineart]
        Selects the scan mode (e.g., lineart, monochrome, or color).
    --resolution 50..600dpi (in steps of 1) [600]
        Sets the resolution of the scanned image.
  Geometry:
    --page-width 0..221.121mm (in steps of 0.0211639) [215.872]
        Specifies the width of the media.  Required for automatic centering of
        sheet-fed scans.
    --page-height 0..876.695mm (in steps of 0.0211639) [279.364]
        Specifies the height of the media.
    -l 0..215.872mm (in steps of 0.0211639) [0]
        Top-left x position of scan area.
    -t 0..279.364mm (in steps of 0.0211639) [0]
        Top-left y position of scan area.
    -x 0..215.872mm (in steps of 0.0211639) [215.872]
        Width of scan-area.
    -y 0..279.364mm (in steps of 0.0211639) [279.364]
        Height of scan-area.
  Enhancement:
    --brightness -127..127 (in steps of 1) [0]
        Controls the brightness of the acquired image.
    --contrast -127..127 (in steps of 1) [0]
        Controls the contrast of the acquired image.
    --threshold 0..255 (in steps of 1) [0]
        Select minimum-brightness to get a white point
    --rif[=(yes|no)] [no]
        Reverse image format
    --ht-type Default|Dither|Diffusion [inactive]
        Control type of halftone filter
    --ht-pattern 0..3 (in steps of 1) [inactive]
        Control pattern of halftone filter
    --emphasis -128..127 (in steps of 1) [0]
        Negative to smooth or positive to sharpen image
    --variance 0..255 (in steps of 1) [0]
        Set SDTC variance rate (sensitivity), 0 equals 127
  Advanced:
    --ald[=(yes|no)] [no] [advanced]
        Scanner detects paper lower edge. May confuse some frontends.
[... 46 lines cut ...]
    --b5-paper[=(yes|no)] [no] [hardware]
        B5 paper detected
    --page-loaded[=(yes|no)] [yes] [hardware]
        Page loaded
    --omr-df[=(yes|no)] [no] [hardware]
        OMR or double feed detected
    --cover-open[=(yes|no)] [no] [hardware]
        Cover open
    --power-save[=(yes|no)] [no] [hardware]
        Scanner in power saving mode
    --email[=(yes|no)] [no] [hardware]
        Email button
    --manual-feed[=(yes|no)] [no] [hardware]
        Manual feed selected
    --scan[=(yes|no)] [no] [hardware]
        Scan button
    --function <int> [1] [hardware]
        Function character on screen
    --double-feed[=(yes|no)] [no] [hardware]
        Double feed detected
    --error-code <int> [32] [hardware]
        Hardware error code
    --skew-angle <int> [0] [hardware]
        Requires black background for scanning
```

cyanfish, 2023-04-09:
Can you try this version? It should set the page-width and page-height options too. [naps2-7.0b7-testsane.deb.gz](https://github.com/cyanfish/naps2/files/11186138/naps2-7.0b7-testsane.deb.gz)

xM8WVqaG, 2023-04-09:
Thanks for the rapid turn around, however, this doesn't appear to have worked. The test cases above in this PR all behave as they did in the previous version.

Just for your peace of mind, I definitely removed the previous version before testing the new version!

```text
$ sudo apt remove naps2
...
Removing naps2 (7.0.7) ...
$ sudo apt install ./naps2-7.0b7-testsane.deb
```

I did notice the new install `Depends` & `Recommends` which did seem to work!

```diff
8,10c8,11
< Download-Size: 18.2 MB
< APT-Manual-Installed: yes
< APT-Sources: /REDACTED/naps2-7.0b7-linux-x64.deb
---
> Depends: libsane1
> Recommends: sane-airscan
> Download-Size: 16.2 MB
> APT-Sources: /REDACTED/naps2-7.0b7-testsane.deb
```

xM8WVqaG, 2023-04-10:
I tried doing a quick `strace` to see if I could find it doing anything obviously wrong but no dice. It doesn't appear to be writing any exception to the `errorlog.txt`either.

Although I did notice it spends a bit of time just enumerating all other processes at startup before it even begins which is weird!

```shell
strace -f -e trace=open,openat -u naps --output log-open.txt naps2 console -o "test.pdf" -v
```

[log-open.txt](https://github.com/cyanfish/naps2/files/11190389/log-open.txt)

cyanfish, 2023-04-10:
I actually went and ordered a FI-6130Z as there are multiple issues with this series that I think are fixable if I have a physical scanner. I'll come back to this when that arrives.

xM8WVqaG, 2023-04-19:
Thanks, I really appreciate your support. I've made a token financial contribution towards the new hardware.

To be transparent, I made the contribution with the understanding that you may not be able to support my hardware on Linux/SANE. Even if you're not able to resolve my issue here, at least you won't be _completely_ out of pocket!

And if you somehow got the hardware for less than I sent, enjoy dinner on me!

cyanfish, 2023-04-23:
Thank you very much! Please try [7.0b8](https://www.naps2.com/download) and see if that fixes the issue.

xM8WVqaG, 2023-05-01:
Tested with the flatpak installer, it looks like this mostly fixed for me

* Scanning with "Paper source" set to "Duplex" with "Page size" set to "A4" no longer hangs :heavy_check_mark: 
* Page size is now _almost_ correctly being communicated
  - Setting "Page size" to "A4" still truncates at "US Letter" ❌
  - Setting "Page size" to 2 inches^2 now only scans a portion of the document :heavy_check_mark: 

Should I open another ticket for the page truncation issue?

xM8WVqaG, 2024-02-04:
The original hanging issue was fixed in `7.0b8`, closing in favour of #281 which captures the A4 truncation issue.

closed 2024-02-04 by xM8WVqaG