cyanfish/naps2 #131: Brother DS-740 - no duplex

Laserlicht, 2023-04-13:
**Describe the bug**
DS-740 scan only in simplex. Don't matter whats configured.

I get only one page.

**To Reproduce**
Steps to reproduce the behavior:
1. Set scanner to duplex
2. scan

**Expected behavior**
Get two pages.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Linux Mint Cinnamon
 - Version 21.1
 - Package: Deb

**Additional context**
`Output of scanimage -A`:
```
Output format is not set, using pnm as a default.
scanimage: rounded value of br-x from 215.9 to 215.88
scanimage: rounded value of br-y from 355.6 to 355.567

All options specific to device `brother5:bus4;dev3':
  :
    --mode 24bit Color[Fast]|Black & White|True Gray|Gray[Error Diffusion] [24bit Color[Fast]]
        Select the scan mode
    --resolution 100|150|200|300|400|600|1200dpi [100]
        Sets the resolution of the scanned image.
    --source Automatic Document Feeder(left aligned)|Automatic Document Feeder(left aligned,Duplex) [Automatic Document Feeder(left aligned)]
        Selects the scan source (such as a document-feeder).
    --brightness -50..50% (in steps of 1) [inactive]
        Controls the brightness of the acquired image.
    --contrast -50..50% (in steps of 1) [inactive]
        Controls the contrast of the acquired image.
    --MultifeedDetection[=(yes|no)] [inactive]
        
    --AutoDocumentSize[=(yes|no)] [no] [advanced]
        
    --AutoDeskew[=(yes|no)] [no] [advanced]
        
    --SkipBlankPage[=(yes|no)] [inactive]
        
    --SkipBlankPageSensitivity 0..100% (in steps of 1) [inactive]
        
  Geometry:
    -l 0..215.9mm (in steps of 0.0999908) [0]
        Top-left x position of scan area.
    -t 0..355.6mm (in steps of 0.0999908) [0]
        Top-left y position of scan area.
    -x 0..215.9mm (in steps of 0.0999908) [215.88]
        Width of scan-area.
    -y 0..355.6mm (in steps of 0.0999908) [355.567]
        Height of scan-area.
```

cyanfish, 2023-04-23:
Please try [7.0b8](https://www.naps2.com/download) and see if that fixes the issue.

Laserlicht, 2023-04-23:
@cyanfish Thanks for your support.

Still only simplex :(

debuglog.txt
```
2023-04-23 13:45:42.5321 50727 Option: 2 mode Scan mode Select the scan mode String None 30 SoftSelect, SoftDetect StringList 24bit Color[Fast],Black & White,True Gray,Gray[Error Diffusion] 
2023-04-23 13:45:42.6030 50727 Option: 3 resolution Scan resolution Sets the resolution of the scanned image. Int Dpi 4 SoftSelect, SoftDetect WordList 100,150,200,300,400,600,1200 
2023-04-23 13:45:42.6030 50727 Option: 4 source Scan source Selects the scan source (such as a document-feeder). String None 64 SoftSelect, SoftDetect StringList Automatic Document Feeder(left aligned),Automatic Document Feeder(left aligned,Duplex) 
2023-04-23 13:45:42.6030 50727 Option: 5 brightness Brightness Controls the brightness of the acquired image. Fixed Percent 4 SoftSelect, SoftDetect, Inactive Range -50-50/1 
2023-04-23 13:45:42.6030 50727 Option: 6 contrast Contrast Controls the contrast of the acquired image. Fixed Percent 4 SoftSelect, SoftDetect, Inactive Range -50-50/1 
2023-04-23 13:45:42.6030 50727 Option: 7 MultifeedDetection Multifeed Detection  Bool None 4 SoftSelect, SoftDetect, Inactive, Advanced None  
2023-04-23 13:45:42.6030 50727 Option: 8 AutoDocumentSize Auto Document Size  Bool None 4 SoftSelect, SoftDetect, Advanced None  
2023-04-23 13:45:42.6030 50727 Option: 9 AutoDeskew Auto Deskew  Bool None 4 SoftSelect, SoftDetect, Advanced None  
2023-04-23 13:45:42.6030 50727 Option: 10 SkipBlankPage Skip Blank Page  Bool None 4 SoftSelect, SoftDetect, Inactive, Advanced None  
2023-04-23 13:45:42.6030 50727 Option: 11 SkipBlankPageSensitivity Skip Blank Page sensitivity  Fixed Percent 4 SoftSelect, SoftDetect, Inactive, Advanced Range 0-100/1 
2023-04-23 13:45:42.6030 50727 Option: 13 tl-x Top-left x Top-left x position of scan area. Fixed Mm 4 SoftSelect, SoftDetect Range 0-215,90000762951095/0,09999237048905166 
2023-04-23 13:45:42.6030 50727 Option: 14 tl-y Top-left y Top-left y position of scan area. Fixed Mm 4 SoftSelect, SoftDetect Range 0-355,6/0,09999237048905166 
2023-04-23 13:45:42.6030 50727 Option: 15 br-x Bottom-right x Bottom-right x position of scan area. Fixed Mm 4 SoftSelect, SoftDetect Range 0-215,90000762951095/0,09999237048905166 
2023-04-23 13:45:42.6030 50727 Option: 16 br-y Bottom-right y Bottom-right y position of scan area. Fixed Mm 4 SoftSelect, SoftDetect Range 0-355,6/0,09999237048905166 
2023-04-23 13:45:42.6030 50727 Maybe setting source 
2023-04-23 13:45:42.6030 50727 Maybe setting adf_mode 
2023-04-23 13:45:42.6030 50727 Maybe setting adf-mode 
2023-04-23 13:45:42.6030 50727 Maybe setting mode 
2023-04-23 13:45:42.6071 50727 Maybe setting resolution 
2023-04-23 13:45:42.6071 50727 Setting resolution to 300 
2023-04-23 13:45:42.6071 50727 Maybe setting tl-x 
2023-04-23 13:45:42.6071 50727 Setting tl-x to 0 
2023-04-23 13:45:42.6071 50727 Maybe setting tl-y 
2023-04-23 13:45:42.6071 50727 Setting tl-y to 0 
2023-04-23 13:45:42.6071 50727 Maybe setting br-x 
2023-04-23 13:45:42.6071 50727 Setting br-x to 210 
2023-04-23 13:45:42.6071 50727 Maybe setting br-y 
2023-04-23 13:45:42.6071 50727 Setting br-y to 297 
```

cyanfish, 2023-04-23:
Oops, that's my bad, I didn't actually include the potential fix in that build. Can you try this version instead? [naps2-7.0b8-testsane.deb.gz](https://github.com/cyanfish/naps2/files/11305069/naps2-7.0b8-testsane.deb.gz)

If that works, could you also check that Black & White and Grayscale options work too?

Laserlicht, 2023-04-23:
Thank you very much! For this great project.

Now everything is working fine. I've also tested B&W and Grayscale.

closed 2023-04-23 by Laserlicht