cyanfish/naps2 #133: Feature: Integrate Crop With Preview WIndow

EmperorArthur, 2023-04-17:
**Is your feature request related to a problem? Please describe.**
Yes.

Crop always shows as a tiny window compared to preview, and the scanned image is always scaled to the window size.  This makes it difficult to crop large scans.  Plus, the bars in the crop window are not labeled.

**Describe the solution you'd like**

Integrate the crop tool in the "Preview" window.

When crop is selected, the cursor would work the same as it does not for crop.  Similarly the greyed out area would stay the same.  However, instead of 4 bars around the edges, use thick dragable lines instead.

**Describe alternatives you've considered**
Add at least a zoom feature to the crop window.

**Additional context**
"Crop" shows up as a completely separate window in Task manager.  However "Preview" does not.

cyanfish, 2023-04-17:
You can try the crop tool in [version 7 beta](https://www.naps2.com/download), as it has draggable lines. You should also be able to resize the window to fill your screen (and it will persist that the next time you open the crop window).

EmperorArthur, 2023-04-17:
That interface is much better, and worked for what I just used it for.  My concern is people with smaller screens that are not 4k.  I have been forced to use the tool on a laptop monitor. The experience was not great, to put it mildly.

Especially since a bad crop means re-scanning the image.

EmperorArthur, 2023-04-17:
Given this and #39 I wonder if it might be better to use some sort of generic image editor as an embed / library.

cyanfish, 2023-04-17:
You can use Image->Reset, you don't need to rescan.

christianclavet, 2023-04-28:
Nice idea about zooming and panning while editing. I will investigate on how to integrate a view windows that can zoom and pan and perhaps submit a pull request for this, when I'm more accustomed to edit the GUI with this new source.

Alternatively @cyanfish, you could enable the GROW icon on the window border of theses small windows. I've done this and it take the full screen. Congrat about being able to apply the saved size when recreating the floating window!

cyanfish, 2024-04-09:
Closing this as it's mostly resolved, and #312 tracks single-window image editing.

closed 2024-04-09 by cyanfish