cyanfish/naps2 #135: Overwrite (--force) not working with --split option

pucis, 2023-04-20:
**Describe the bug**

I'm running ORC on existing PDF files, and then spliting each page into separate PDF.
If I run the same command second time with --force option, existing pages are not overwritten, but additional files are created.

**To Reproduce**
Running this command:
NAPS2.Console.exe --import "02012023_P_1.pdf" --enableocr --ocrlang "eng+lav" --split --output "pages\02012023_P_1.pdf" --force -n 0 --verbose

There is generated file for each of 10 pages:
02012023_P_1.01.pdf ... 02012023_P_1.10.pdf

Running the same command again, previous files are not overwritten, but created 10 new files
02012023_P_1.11.pdf  ... 02012023_P_1.20.pdf

**Expected behavior**
--force option should overwrite existing files

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Version 7.0b7

willibrandon, 2023-10-22:
I have a possible solution for this issue if you're interested.

We should consider the `ForceOverwrite` option value during `AutomatedScanning` when performing [place holder substitution](https://github.com/cyanfish/naps2/blob/538993c7ace4116f3f8c84832db9c8e7866c47c4/NAPS2.Sdk/ImportExport/Placeholders.cs#L101C2-L101C2) and pass the negation of that value in for parameter `incrementIfExists`.

https://github.com/cyanfish/naps2/blob/538993c7ace4116f3f8c84832db9c8e7866c47c4/NAPS2.Lib/Automation/AutomatedScanning.cs#L588C32-L588C32

Also, the same should be done in `SavePdfOperation` using the value returned from `_overwritePrompt.ConfirmOverwrite(fileName) == OverwriteResponse.Yes`.

https://github.com/cyanfish/naps2/blob/538993c7ace4116f3f8c84832db9c8e7866c47c4/NAPS2.Lib/Automation/AutomatedScanning.cs#L588C32-L588C32

cyanfish, 2026-10-25:
Fixed in [8.6.0](https://www.naps2.com/download).

closed 2026-10-25 by cyanfish