cyanfish/naps2 #139: Question: Source code, where is the GUI files(.resx)?

christianclavet, 2023-04-28:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
I'm looking to create pull request for porting my ideas from my fork to your new code. But can't find the .resx files to edit the GUI. 

**Describe the solution you'd like**
A clear and concise description of what you want to happen.
Hints or explanation on how to edit the GUI elements of the new NAPS

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.
None so far

**Additional context**
Add any other context or screenshots about the feature request here.

cyanfish, 2023-04-28:
NAPS2 7.x uses [Eto](https://github.com/picoe/Eto) as a cross-platform GUI framework, with a [custom](https://github.com/cyanfish/naps2/tree/master/NAPS2.Lib/EtoForms/Layout) layout system that's conceptually similar to Eto's [DynamicLayout](https://github.com/picoe/Eto/wiki/DynamicLayout) but works a bit differently. All cross-platform GUI code is in [NAPS2.Lib/EtoForms](https://github.com/cyanfish/naps2/tree/master/NAPS2.Lib/EtoForms), while platform-specific code is in the corresponding project, e.g. [NAPS2.Lib.WinForms/EtoForms](https://github.com/cyanfish/naps2/tree/master/NAPS2.Lib.WinForms/EtoForms).

The best way to get a feel for it is probably to look at the BuildLayout methods in various [forms](https://github.com/cyanfish/naps2/tree/master/NAPS2.Lib/EtoForms/Ui) (e.g. [ProfilesForm](https://github.com/cyanfish/naps2/blob/365afce2a4915abb18fc918e6194f27a1ae30f13/NAPS2.Lib/EtoForms/Ui/ProfilesForm.cs#L122) or [EditProfileForm](https://github.com/cyanfish/naps2/blob/365afce2a4915abb18fc918e6194f27a1ae30f13/NAPS2.Lib/EtoForms/Ui/EditProfileForm.cs#L78)). All you do is declare the visual relationships of controls (in rows and columns) and control position, form sizing, etc. are generally handled automatically - which is important as the exact dimensions tend to differ across Windows/Mac/Linux.

Some things that aren't that simple are handled ad hoc - for example, [DesktopForm](https://github.com/cyanfish/naps2/blob/master/NAPS2.Lib/EtoForms/Ui/DesktopForm.cs) has subclasses for WinForms/Mac/Gtk as the desktop is different on each platform (particularly Mac). And list views (as used for the main image list, profiles list, and ocr language downloads) have a [custom abstraction](https://github.com/cyanfish/naps2/blob/master/NAPS2.Lib/EtoForms/Widgets/IListView.cs) that's implemented on each platform.

NAPS2 by default builds 64 bit exes and should only be limited by system memory (which was also true in 6.1.2, with the exception of scanning large images with TWAIN, which is fixed in 7.x).

christianclavet, 2023-04-29:
Thanks, I've been able to understand to build your last version. I will have a look at ETO to understand how it work. Thanks for all the information you provided. 

Once understood, I should be able to create theses pull requests and you'll decide if take them or not, no pressure here.
I did not practice programming for years until I decided to modify the 6.1.2 source for my need, and it helped me a lot improve and better understand (I'm much more a designer than a programmer). Thanks for a the help!

christianclavet, 2023-05-02:
Ok. 

Been able to found some of the stuff there in the LIB project in the UI section. But it will not be easy, as the items of the list view I was using are not exposed. Will have to tweak directly in the class of the thumbnail controller to display the thumbnails info. 

Might take a while before I can post my first pull request...

You made quite a lot of the changes there!👍 Must learn this new code!😊

cyanfish, 2023-05-02:
What specifically are you trying to do?

christianclavet, 2023-05-03:
Hi! I will check more of the new code before doing it, but basically, only use the icon text of the listview and give it a number, and how many picture you had (ex: 1/80). 

I will have some more time in the weekend to learn more about the code and expose the proper stuff to be able to change it. I would like to create a PR for each feature I would like to bring to this new version of Naps. So there will be a Pr for each. 

But I'll start with the simple stuff first. The first thing I want is to display the numbering on the pictures, and I only need to expose some things and should be simple.

If I'm doing this, is there things to do on the Mac and Linux side?
Thanks again for sharing this project!

cyanfish, 2023-05-04:
This should be mostly doable by making [ImageListViewBehavior](https://github.com/cyanfish/naps2/blob/master/NAPS2.Lib/EtoForms/Widgets/ImageListViewBehavior.cs) more like [ProfileListViewBehavior](https://github.com/cyanfish/naps2/blob/master/NAPS2.Lib/EtoForms/Widgets/ProfileListViewBehavior.cs), namely setting ShowLabels and implementing GetLabel.

A couple caveats:
- Label rendering may not work perfectly (especially on Mac/Linux) with full-size images.
- Label rendering may not account for labels changing as images are reordered/added, the assumption is probably that labels are unchanging. This may require changes to [WinFormsListView](https://github.com/cyanfish/naps2/blob/master/NAPS2.Lib.WinForms/EtoForms/WinForms/WinFormsListView.cs) (and the Mac/Linux equivalents).

In principle I'd be willing to merge a PR with this with a couple notes:
- The behavior should be behind a config flag ([example](https://github.com/cyanfish/naps2/commit/c5f1d82d0954c04625ac15e0c8648c3d773f8606)).
- There would be no way in the UI to set this value yet. Eventually I plan to add a proper settings dialog but until then you'd have to edit the AppData config.xml file manually.
- I'm okay with it not fully working/tested on Mac/Linux given the above.

And also if you want to do other UI changes, using a config flag would apply to those too as I don't want to make the UI "noisier" for the average user.

christianclavet, 2023-05-06:
Hi, Started to look at this. Wow! Thanks for the hints! Yes! for the enabler that perfect with me, will add one at each PR.👍
Reading at the code, the Listview seem to be implemented fully in Naps2 and you defined those behavior re-using it for the Image Listview, Profiles, and OCR... That's really impressive! 👌

- I've been able to set the config flag (not yet used in the code, since the "_displayName_" is not yet defined.)
[image]

- Done the recommendations for the _ImageListBehavior_ and implemented _ShowLabels_ as in the _ProfileListViewBehavior_.
[image]
and added this in _UiImage.cs_
[image]

- Done a quick check and here is the result: It make the pictures square
[image]
I know the pictures for the Profiles are Square, and not sure why it's happening. Since the content of the labels are empty, it should not affect the display. Do you know what is happening there?

cyanfish, 2023-05-06:
Ah yes, I'd forgotten about that. The issue is that turning ShowLabels on turns [UseCustomRendering](https://github.com/cyanfish/naps2/blob/6f4787464b4da1e7eb5d3a2259fab0a955eed40d/NAPS2.Lib.WinForms/EtoForms/WinForms/WinFormsListView.cs#L72) off which needs to be on for rendering images with varying dimensions. I would recommend:
1. Add a `UseCustomRendering` ListViewBehavior property (to replace the property on WinFormsListView).
2. Set it to true for ImageListViewBehavior.
3. Tweak the [rendering](https://github.com/cyanfish/naps2/blob/6f4787464b4da1e7eb5d3a2259fab0a955eed40d/NAPS2.Lib.WinForms/EtoForms/WinForms/WinFormsListView.cs#L74) to draw the label when ShowLabels is true (might be a bit tricky to get right).

christianclavet, 2023-05-07:
Ok. Found out it was using Custom Rendering, and it changed to the style of the others Listview...
The implementation is written in "WinFormListView.cs" when doing custom rendering.
Here is the result:
[image]
This can only be enabled if the flag of the config is set to true as you requested.
I'll upload this on my fork and create my first PR for this first one! One thing remaining is to draw the text also on the Linux and Mac side... (Can't really do it)

closed 2023-05-07 by christianclavet