cyanfish/naps2 #142: Enabling 'Crop to page size' crops the images to nothing

xM8WVqaG, 2023-05-01:
**Describe the bug**
A clear and concise description of what the bug is.

After enabling "Crop to page size" the output is not unusable.

* The thumbnails and preview are broken. They look kinda like FFT-plots of the image(?) although I expect that might not be relevant.
* Saving the output creates 1x1 images and plain PDFs.

  ```text
  $ file crop-to-page-size.1.jpg
  crop-to-page-size.1.jpg: JPEG image data, JFIF standard 1.01, aspect ratio, density 1x1, segment length 16, baseline, precision 8, 1x1, components 3
  ```

**To Reproduce**
Steps to reproduce the behavior:

Under "Profile" > "Advanced" > "Compatibility", enable "Crop to page size".

[image]

The two images on the left are scanned in colour. The two on the right are the same document in greyscale.

[image]

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Ubuntu 22.04

```text
NAPS2 - Not Another PDF Scanner - Scan documents to PDF and more, as simply as
possible.

          ID: com.naps2.Naps2
         Ref: app/com.naps2.Naps2/x86_64/beta
        Arch: x86_64
      Branch: beta
     Version: 7.0b8
     License: GPL-2.0-or-later
      Origin: naps2-origin
  Collection: 
Installation: user
   Installed: 72.7 MB
     Runtime: org.freedesktop.Platform/x86_64/22.08
         Sdk: org.freedesktop.Sdk/x86_64/22.08

      Commit: 27e9023098c424f3aca01b33100722e169b76970db97671f4c83e1356ecedb1b
      Parent: 61dca37fe0ff53c2e876a59e6bca039366523d7cacc00e08c793cef396fa774b
     Subject: Export com.naps2.Naps2
        Date: 2023-04-22 22:54:49 +0000
```

**Additional context**
Add any other context about the problem here.

christianclavet, 2023-05-13:
What was your paper size defined in the profile when you've done the test and what size was the real paper? Never user this feature before.

xM8WVqaG, 2023-05-13:
I just checked - this was the last thing I did with NAPS so the profile was still configured - the paper size was "US Legal" and the paper in real life was A4 (everything I have is A4). Although I have another issue where all NAPS scans I make truncate to "US Letter".

I haven't used this feature via SANE before, but I have used a similar feature on Windows (probably) via TWAIN. It's quite useful for my scanner because it has a feeder with quite a wide slot so the paper guides don't keep paper perfectly straight when there is only one or two sheets.

cyanfish, 2023-07-09:
This should be fixed in [7.0b9](https://www.naps2.com/download).

closed 2023-07-09 by cyanfish