cyanfish/naps2 #145: Last profile used for scanning is saved as standard profile

NextTherapist, 2023-05-09:
**Describe the bug**
Create at least two scanning profiles.
Declare one of them as the standard profile.
Scan a document via button (arrow down) or via F2 or F3 with the OTHER profile.
And voilà: the used profile now is the standard profile, just by using it one time!

**Expected behavior**
The standard profile should stay the standard profile regardless which profile was used last.

**Desktop (please complete the following information):**
 - OS: [e.g. Windows 10
 - Version 7.0.8

cyanfish, 2023-05-15:
This is intended behavior but I do want to add an option to change it.

beatrupp, 2023-07-26:
How about the case when you only have one profile? Couldn't this be automatically preselected and used (saving one user interaction click)?

cyanfish, 2024-01-15:
In [NAPS2 7.3.0](https://www.naps2.com/download) I've added a "Settings" window with a new option:
- "Scan" menu changes default profile
You can uncheck this to stop it from changing.

closed 2024-01-15 by cyanfish