cyanfish/naps2 #151: [question] Will the sdk support .net maui?

DesmondMilez, 2023-06-09:
Hello. 
I am currently working on a .net maui desktop app that target both windows and macOS. I need to work with installed scanners on the device but as of today .NET MAUI targets MacCatalyst which still doesn't support AppKit namespace. I saw the source code of the SDK project and saw dependencies to ImageCaptureCore which won't compile on .NET MAUI. 
My question is if this SDK will have any support of MAUI (maybe using SANE) or no?

Thanks

cyanfish, 2023-06-09:
The SDK should support SANE (and eventually ESCL) with the net6 build. But the other issue is that on Mac it currently depends on the [NSImage](https://github.com/cyanfish/naps2/tree/master/NAPS2.Images.Mac) type - eventually I plan to add support for ImageSharp and SkiaSharp but that won't be in the first release.

cyanfish, 2023-09-05:
As of NAPS2.Sdk [0.1.0-alpha04](https://www.nuget.org/packages/NAPS2.Sdk) the SDK should, on Mac, support both SANE and ESCL with the net6 build. It also should be usable on all platforms with the [NAPS2.Images.ImageSharp](https://www.nuget.org/packages/NAPS2.Images.ImageSharp/) package.

closed 2023-09-05 by cyanfish