cyanfish/naps2 #152: Barcode placeholder in filenames

cyanfish, 2023-06-09:


raylutz, 2023-10-03:
I copied this from a discussion now that I see this issue raised.

I am looking for a batch scanning function where the text from code39 barcode on Patch-T page is used as a placeholder in file name.
So if Barcode says "AB0001" then I can use that as placeholder like $(barcode) in the generated filename, like
$(barcode)-$(nnnnnn).pdf
So that each page scanned in duplex will be prefixed by barcode and followed by 6 digit number, like
AB0001-0000001.pdf

Each time a new batch is started, then the second number would restart, but this could be an option.

would be the first document scanned of two pages (each side) in pdf format, hopefully using ccitt internally.

I can probably find funding if someone can implement this at reasonable fixed cost.

Thanks.

GoodFriendJob, 2023-10-18:
I think you can generate the file name with the current timestamp. If it's acceptable clue, I can update the code.

GoodFriendJob, 2024-05-24:
> Is this feature planned to be integrated soon? thanks

Sure. Please get me a chance to complete asap.