cyanfish/naps2 #153: Usability request: Buttons per profile

folknor, 2023-06-10:
**Is your feature request related to a problem? Please describe.**
I have installed naps2 on at least 12 customer PCs in different companies, and every one of them has three problems:
1. Hitting the small arrow next to the Scan button with their mouse (they all need multiple profiles, every person needs it)
2. Remembering that hitting the Scan button uses the last used profile (ref https://github.com/cyanfish/naps2/issues/145)
3. Noone understands the keyboard shortcuts, except one person, and even if they do they can't remember which scan method + scanner is which keyboard shortcut anyway without opening the tiny menu

**Describe the solution you'd like**
Here is a mockup of one potential solution:
[image]

Either the buttons would:
1. Grow downwards and a scrollbar would appear if needed, or
2. Grow down-and-right in multiple columns, and "eat" more of the document area if needed

I would prefer #2.

Also, I forgot to add keyboard shortcuts visible in the mockup UI.

I propose that the application is extended in two ways:
1. You add the ability to configure a custom icon per device
2. You add custom icons for Glassplate, Feeder, and "Feeder two-sided"

And then, every profile icon in the new sidebar would be a combination icon, something like this bad mockup:
[image]

The tooltip for the button should include at least scan device, scan source (feeder, glass plate, etc), and DPI.

Does this make sense?

Thank you for reading! :100: 

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

xZise, 2023-06-11:
Hi, maybe the first issue could be improved more easily (at least with WinForms) by not splitting the button so the "down arrow"-action can be reached by clicking anywhere on it (and it is just there to drop down).

cyanfish, 2024-01-15:
In [NAPS2 7.3.0](https://www.naps2.com/download) I've added a "Settings" window with a few options related to this:
- Show "Profiles" toolbar (adds one-click shortcuts for each profile)
- "Scan" menu changes default profile (uncheck to keep the same default profile)
- "Scan" button default action (change to "Always Prompt" to show the dropdown instead of one-click scanning)

I've not done custom icons but feel free to open a new issue for that.

closed 2024-01-15 by cyanfish

folknor, 2024-01-31:
Excellent work, thank you!