cyanfish/naps2 #156: Recently scanner operating panel pops up and asks for Start/Stop

NextTherapist, 2023-06-12:
**Describe the bug**
When scanning with my NAPS profiles, since today the TWAIN scanner driver (Canon CaptureOnTouch 2.8) shows up with a little window and presents two buttons: Start and Stop. He never did that before. In the past I just pressed the scanning button in NAPS and some time later I had a complete PDF without any questions from the twain driver.

How could I avoid this behaviour?

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Version 7.0b8

cyanfish, 2023-06-14:
You can try changing TWAIN Implementation in the advanced profile settings, but it probably won't make a difference as this sounds related to the driver. The main thing I would recommend is to try reinstalling that driver.

NextTherapist, 2023-07-02:
Reinstalling the driver unfortunately did not help. I had to go into the NAPS profile, choose the option "Scanner dialogue" and then change the behaviour in the scanner's own CaptureOnTouch profile. Very strange.

cyanfish, 2023-07-08:
Interesting, I will keep that in mind in case anyone else has the same problem.

closed 2023-07-08 by cyanfish