cyanfish/naps2 #157: Profile Settings and Preview Brightness, Contrast, Black & White changes not saved

JosepBalague, 2023-06-13:
**Describe the bug**
In Profile Settings if I set Brightness and/or Contrast to any value, scanning result is always the same.
Now, both Brightness and Contrast seems useless options.

**To Reproduce**
1. Create Profile Settings (B=0, C=0)
2. Scan any page and save as PNG
3. Modify Profile Settings (B=-1000, C=-1000)
4. Scan any page and save as PNG
5. See resulting PNGs are visually the same and it shouldn't (second PNG should be diferent from first)

**Expected behavior**
Second PNG should be dark image. Profile Settings should be applied to the resulting image.

**Screenshots**
[image]

**Desktop (please complete the following information):**
- Microsoft Windows [Version 10.0.22621.1778]
- Naps2 [7.0b8]
- Paint.net 5.0.6

**Additional context**
n/a

cyanfish, 2023-06-13:
This looks like the same issue as #107.

You can select "Apply brightness/constrast after scan" under Advanced in your profile settings to bypass the scanner's brightness/contrast capabilities (which sometimes don't work).

At some point I want to make this the default behavior so it doesn't need a separate option.

cyanfish, 2023-06-13:
Actually let me revise that - it looks like the same problem as #158 might be affecting this too even with that setting.

JosepBalague, 2023-06-13:
> 

Hi Ben,

Thank you for responding so quickly.

This is my Profile Settings:

[image]

Saved PNG should be dark image but still the same as To Reproduce, Step 2 image.

Thank you, I will wait for your comments.

cyanfish, 2023-06-18:
Closing as the same problem as #158.

closed 2023-06-18 by cyanfish